mavlink/qgroundcontrol #1991: Waypoint reordering results in missing icons

LorenzMeier, 2015-10-14:
[image]

The up / down controls seem to lead to issues on the map regarding the icons.

DonLakeFlyer, 2015-10-14:
Yeah, I've seen that. It some deep dark bug in Qt. What do you think about Up/Down? Doesn't seem like critical functionality to me. I'd rather remove than dig into the bowels of Qt to fix.

LorenzMeier, 2015-10-14:
Yeah, remove it. Making them movable is by far more relevant.

LorenzMeier, 2015-10-14:
And I would not bring it back. Too confusing from a user perspective. Users move stuff on the map, not in the list.

LorenzMeier, 2015-10-14:
Put the delete into an X icon in the list.

DonLakeFlyer, 2015-10-14:
Yeah, that was the direction I was heading

closed 2015-12-24 by DonLakeFlyer