mavlink/qgroundcontrol #1996: Waypoint manager false warnings: Connect logic should depend on first heartbeat, not on connection open.

LorenzMeier, 2015-10-15:
Because the connect is based on link connect and not first heartbeat the waypoint manager runs into a timeout as it incorrectly assumes it can talk to the vehicle once the link is up. This happens when connecting via USB right after Pixhawk is enumerated for the first time on the bus after plugging in. In that state the board is still initialising.

If my guess is wrong and its already waiting on the heartbeat then the timeout likely needs to be increased.
[image]

```
Filter rules "qt.network.ssl.warning=false
MissionManagerLog.debug=true
"
Saved files location "/Users/lorenzmeier/Documents/QGroundControl"
Load setting "Sim"
"9px;" "12px;" "19px;"
0
Orphaned log file count 0
Temp log "/var/folders/zc/zsk5myvj69b0hfgz2rbj8_bh0000gp/T/FlightData023997.mavlink"
Unknown element in XML:  "board"
PX4ParameterLoaderLog: Invalid default value, name: "COM_EF_TIME"  type: 6  default: QVariant(float, 10)  error: "Value must be within 0 and 7"
PX4ParameterLoaderLog: Invalid default value, name: "NAV_DLL_CH_LAT"  type: 5  default: QVariant(int, -266072120)  error: "Value must be within 0 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "NAV_AH_LAT"  type: 5  default: QVariant(int, -265847810)  error: "Value must be within 0 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "MAV_TYPE"  type: "INT32"  default: "MAV_TYPE_FIXED_WING"  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "PE_MAG_NOISE"  type: 6  default: QVariant(float, 0.05)  error: "Value must be within 0.1 and 10"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC8_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC8_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC8_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC9_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC9_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC9_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC10_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC10_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC10_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC11_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC11_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC11_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC12_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC12_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC12_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC13_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC13_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC13_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC14_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC14_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC14_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC15_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC15_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC15_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC16_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC16_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC16_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC17_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC17_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC17_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC18_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC18_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
[... cut ...]
MissionManagerLog: _retrySequence ackType: "MISSION_COUNT" _retryCount 12
MissionManagerLog: Invalid ack sequence _retryAck:expectedAck "No Ack" "MISSION_COUNT"
MissionManagerLog: _retrySequence ackType: "MISSION_COUNT" _retryCount 13
MissionManagerLog: Invalid ack sequence _retryAck:expectedAck "No Ack" "MISSION_COUNT"
MissionManagerLog: _retrySequence ackType: "MISSION_COUNT" _retryCount 14
MissionManagerLog: Invalid ack sequence _retryAck:expectedAck "No Ack" "MISSION_COUNT"
MissionManagerLog: _retrySequence ackType: "MISSION_COUNT" _retryCount 15
```

DonLakeFlyer, 2015-10-15:
Nothing really happens on connect other than addiing the link to the LinkManager. The first heartbeat comes through MAVLinkProtocol, which in turn creates the Vehicle, which in turn creates the MissionManager and ParameterLoader. They both in turn start their protocol sequence.

For some reason the mission items are being requested three times at boot. You didn't manually request them right? You can see from the log three "MissionManagerLog: requestMissionItems read sequence". Followed by the protocol sequence. The first two times it works fine. On the third one QGC asks for item 1, and the firmware sends back item 0:

```
MissionManagerLog: _requestNextMissionItem sequenceNumber: 1
Unknown message from system: 1 message: 39
MissionManagerLog: _handleMissionItem sequenceNumber: 0
MissionManagerLog: _handleMissionItem mission item received out of sequence expected:actual 1 0
MissionManagerLog: _retrySequence ackType: "MISSION_ITEM" _retryCount 0
MissionManagerLog: _retryRead
MissionManagerLog: Invalid ack sequence _retryAck:expectedAck "MISSION_COUNT" "MISSION_ITEM"
MissionManagerLog: _retrySequence ackType: "MISSION_ITEM" _retryCount 1
MissionManagerLog: _retryRead
```

This then causes it to retry, by restarting the protocol sequence all over again. It gets a MISSION_COUNT response, but then it times out waiting for MISSION_ITEM 0 and restart. Then all hell breaks loose because it looks like the MISSION_ITEM was actually their just very delayed so the protocol sequence starts to causes error and that this point it finally hits some sort of bug on the QGC side which I need to look at some more.

Phew! Anyway, the questions are:
1) Why did QGC end up manually requesting the list more than once. These were not in response to retries. These were direct calls to MissionManager::requestMissionItems. This should only happen once when a Vehicle is created, or when the user requests manually.
2) Why on the third try does the firmware respond with an incorrect item? From that point on the firmware side seems very confused. That looks like a bug on the firmware side.
3) For the last retry sequence I need to find the bug in QGC which is causing it to retry too many times.

DonLakeFlyer, 2015-10-15:
For #3 I was wrong. What is happening there is the firmware is sending MISSION_COUNT over and over again. The MissionManager is at idle at that point, so it just ignores then. So no bug there on QGC side.

DonLakeFlyer, 2015-12-03:
Delaying mission requests till after param load fixed this I believe.

closed 2015-12-03 by DonLakeFlyer