mavlink/qgroundcontrol #2001: QGC doesn't offer a way to switch mavlink connection if more than one mavlink devices are present.

pritamghanghas, 2015-10-15:
I have a pixhawk with ArudPilot on it and storm32 connected to it over telem2 port. If I try to connect to it over telemetry radio. It by default connects to component id 71 of the storm32 and doesn't even offer me a way to switch connection to ardupilot. 

The desired behaviour will be correctly identify the main system and connect to it and offer a way to connect to switch to other subsystems. Mission planner does that.

DonLakeFlyer, 2015-10-16:
So in this case you have both of them coming across the same connection link right? Does the storm show up as:
- An additional component id which is part of the ardupilot system. It does not have it's own heartbeat and system id?
or
- A seperate system which has it's own system id and component id

DonLakeFlyer, 2015-12-03:
@pritamghanghas Help with answers?

pritamghanghas, 2015-12-04:
I will check but I think it comes as an additional component id as part of
same system id. The storm32 connects to pixhawk over serial and then the
storm32 library code as part of ardupilot firmware takes care of putting it
on the mavlink connection.

I think it can also work as  a completely standalone mavlink system as well.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier