mavlink/qgroundcontrol #2002: Clean up Windows installer generation and store debugging artifacts independent of packaged installer

NaterGator, 2015-10-15:
First and foremost the installer does not need to [package](https://github.com/mavlink/qgroundcontrol/blob/3ba9c2f869b7b2d799fc56fa8e3fd6307c6b4c6a/QGCSetup.pri#L204) the PDB files for the debug dlls of Qt's plugins. These PDB symbol databases decompress to over 100MB and have absolutely no relevance to the release build (verifiable through Process Explorer's list of loaded DLLs in a running instance of QGroundControl). 

A better approach to the current process of bulk copying everything would likely be to invoke Qt's tool specifically designed for deploying in windows: [windeployqt](http://doc.qt.io/qt-5/windows-deployment.html#the-windows-deployment-tool). This should place less burden on us in maintaining a manual list of used plugins and would cut out the unnecessary debug versions of the plugins that are caught up by QMAKE_COPY_DIR. 

Additionally best practices dictate that we should not include the application PDB in the installer. We could [adopt the github model](https://github.com/blog/1271-how-we-ship-github-for-windows) and upload the stable release PDB symbols to S3 + use [Pepto Symbol](https://github.com/aroben/pepto-symbol) to resolve them. 

Finally, we should probably consider integrating a multiplatform crash handler like [Google Breakpad](https://chromium.googlesource.com/breakpad/breakpad/+/master/docs/getting_started_with_breakpad.md). Adding breakpad support is a bigger undertaking that warrants a separate issue, but would make collecting user crash reports more consistent. 

For now lets move to:
- [ ] Make the Windows deployment steps less brittle
- [ ] Cut down on unnecessary bloat in the release installer

DonLakeFlyer, 2015-10-16:
Do you have experience with windeployqt? Last time I tried using it I couldn't get it to work. It was a while ago, so I don't remember what the problem was.

I was confused on PDB's. I only put the qgroundcontrol.pdb in there for debugging. The other one are accidental from the way the build scripts are written. It's been like that forever.

NaterGator, 2015-10-16:
I'll need to verify appveyor supports windeployqt but it is working on my
build machine. After switching to it the installer drops to about 45MB with
the application PDB and 40MB without.

I'll update tomorrow and create a pull request for windeployqt... Maybe we
open a separate issue and merge the PDB / symbol server discussion with one
about integrating a crash handler.

DonLakeFlyer, 2015-10-16:
Awesome, thanks Nate.

closed 2015-10-18 by NaterGator