mavlink/qgroundcontrol #2006: Tablet Usability

DonLakeFlyer, 2015-10-15:
This is a catch all Issue for capturing Tablet usability issues. We'll discuss below and then I'll pull up works items up here as we make decisions.

DonLakeFlyer, 2015-10-15:
- I think we should create a single workable color palette for QGC. That means getting rid of the Light/Dark. They are nice, but trying to manage two that both work well is a real PITA.
- I think the Vehicle should just be an outline, with a see through middle. That way you can see waypoints that are under it. And probably some sort of dot for the center point.
- The Fly view widgets need work:
 - They need to be way smaller
 - Need ui changes to be visible over satellite map

[image]

LorenzMeier, 2015-10-16:
The light color theme is pointless because the maps are "dark" and the tablet screens are much brighter outdoors. Its also a design nightmare (looks horrible).

LorenzMeier, 2015-10-16:
Don, can you try to start testing with PX4 SITL? The reason I ask is that this will expose the things that are an issue when running on the tablet, such as the odd connect step via UDP.

LorenzMeier, 2015-10-16:
Fly view doesn't have lines between waypoints. I suggest to ditch its map handling completely. As the person operating the vehicle you want the same as on the plan view. You can make the plan view's sidebar into a ribbon that you can minimise and maximise. When it's maximised you can hide all the flight instruments except for a minimalistic HUD + altitude.

LorenzMeier, 2015-10-18:
So here is my long wanted laundry list: The bottom line is that I'm blown away by the new UI. The logic next step to complete this is to consolidate map handling onto a single view and provide PiP for video.

[image]
[image]
[image]

closed 2015-12-03 by DonLakeFlyer