mavlink/qgroundcontrol #2015: Bump Mac OS target to 10.8

dogmaphobic, 2015-10-17:
Adding here so I won't forget. We should bump the target to 10.8 and fix the deprecated Text to Speech functions currently used (which are not working in my setup).

dagoodma, 2015-10-17:
I am using OS X 10.7.5, and the latest version of XCode I am allowed to install has provided __mac os x 10.8 sdk__, so this is fine by my vote.

I am posting here because I wanted to mention that I had to put the following line under the MacBuild{} section of QGCApplication.pro to fix an error when trying to build about not being able to find __mac os x 10.9 sdk__:

        QMAKE_MAC_SDK= macosx10.8

dogmaphobic, 2016-01-13:
I fixed the original issue, Mac OS TTS, a while back. If we have a compelling reason to change the current target (10.6) we can re-open this.

closed 2016-01-13 by dogmaphobic