mavlink/qgroundcontrol #2019: QGC Not Building in Qt 5.5.1 on Mac OS X

dagoodma, 2015-10-17:
I just installed Qt 5.5.1 tonight on Mac OS X 10.7.5. I updated QGC to latest commit 92cc42a14756860334d4988fc99b1b86908e699a. When trying to build, I see this:

	:-1: error: No rule to make target '/Users/dagoodma/Documents/ASL/dubins_area_coverage/qgroundcontrol/build/build-debug/obj/libQGeoServiceProviderFactoryQGC.a', needed by 'debug/qgroundcontrol.app/Contents/MacOS/qgroundcontrol'.  Stop.

I see that QGC developers are still trying to get it working with Qt 5.5.1. How can I help?

PS. Here is a link to the Makefile: http://m4l3.com/upload/Makefile.txt

Qt Creator 3.5.1 (opensource)
Based on Qt 5.5.1 (Clang 6.0 (Apple), 64 bit)
Built on Oct 13 2015 07:35:31
From revision e548635a24

dogmaphobic, 2015-10-17:
I'm having no issues building on Mac OS, though I'm running 10.10.5. In fact, I was about to target 10.8 as minimum as I was having some other runtime issues. 10.7.5 is a bit on the old end of things. What version of Xcode are you running? What hardware?

dogmaphobic, 2015-10-17:
Also, are you building from within Qt Creator or from Xcode? It doesn't appear you are running a ghost build. That looks like you are either running from Xcode or changed the default build in Qt Creator.

dogmaphobic, 2015-10-17:
One more thing, did you rename your makefile to .txt when you uploaded? You should have 3 separate makefiles:
```
-rw-r--r--  1 gus  staff    41902 Oct 17 03:23 Makefile
-rw-r--r--  1 gus  staff  6784140 Oct 17 03:23 Makefile.QGCApplication
-rw-r--r--  1 gus  staff    94275 Oct 17 03:23 Makefile.QGCLocationPlugin
```

dagoodma, 2015-10-17:
On Mac OS X 10.7.5 the latest version of Xcode I can use is 4.6.3, but it does provide the mac 10.8 sdk. So targetting 10.8 as a minimum works for me. My macbook is a 2008 with intel core 2 duo x64. I don't think many people run anything earlier than that.

I also tried Qt 5.5.0, and I see the same error after doing a clean build. I did change the default build in Qt Creator though. Currently I am using clang 64bit, with clang 4.2 (425.0.28 -- based on llvm32.svn) for target x86_64-apple-darwin11.4.2. Here's the qmake call from my build config:

    qmake /Users/dagoodma/Documents/qgroundcontrol/QGCApplication.pro -r -spec macx-clang CONFIG+=debug CONFIG+=x86_64 CONFIG+=declarative_debug CONFIG+=qml_debug

I am building from Qt Creator:
Qt Creator 3.4.2 (opensource)
Based on Qt 5.5.0 (Clang 6.0 (Apple), 64 bit)
Built on Jun 29 2015 01:54:07
From revision b57ac109a2

I did rename the Makefile to .txt so that it could be read as a text file on the webserver. I do not see Makefiles ending with .QGCApplcation and .QGCLocationPlugin. I think that may be related to my issue. This is all I see in my build directory:
	-rw-r--r--   1 dagoodma  staff  5413631 Oct 17 01:48 Makefile
	drwxr-xr-x   6 dagoodma  staff      204 Oct 16 22:57 build-debug
	drwxr-xr-x   3 dagoodma  staff      102 Oct 16 22:57 debug

where build-debug is a directory with temporary build files, and debug holds qgroundcontrol.app/.

Here is the output when I run the qmake command listed above

	Project MESSAGE: Mac build
	Project MESSAGE: Debug flavor
	Project MESSAGE: BASEDIR /Users/dagoodma/Documents/ASL/dubins_area_coverage/qgroundcontrol DESTDIR /Users/dagoodma/Documents/ASL/dubins_area_coverage/qgroundcontrol/build/debug TARGET QGCApplication OUTPUT /Users/dagoodma/Documents/ASL/dubins_area_coverage/qgroundcontrol/build
	Project WARNING: No MAVLink dialect specified, only common messages supported.
	Project MESSAGE: Skipping support for XBee API (unsupported platform)
	Project MESSAGE: Skipping support for 3DConnexion mice (unsupported platform)
	Project MESSAGE: Skipping support for RT-LAB (unsupported platform)
	Project MESSAGE: Including support for speech output
	Project MESSAGE: Including support for Zeroconf (Bonjour)
	Project MESSAGE: Skipping support for video streaming (GStreamer libraries not installed)
	Project MESSAGE:   You can download it from http://gstreamer.freedesktop.org/data/pkg/osx/
	Project MESSAGE:   Select the devel package and install it (gstreamer-1.0-devel-1.x.x-x86_64.pkg)
	Project MESSAGE:   It will be installed in /Libraries/Frameworks

dagoodma, 2015-10-17:
I also tried Qt 5.4.2, and I have the same issue. What am I doing wrong?

dagoodma, 2015-10-17:
Turns out I had opened the QGCApplication.pro project, and was trying to build that instead of qgroundcontrol.pro. I also had to change **QMAKE_MAC_SDK** to *macosx10.8* in *QGCCommon.pri*.

dagoodma, 2015-10-17:
Another issue.

    :-1: error: framework not found XCTest

I need to disable testing, because I don't have the XCTest.framework for my version of XCode, and I don't have the option of upgrading my XCode since Apple has dropped support for my model of macbook.

I am going to try compiling in Release mode to see if that disable unit testing.

dagoodma, 2015-10-17:
Another issue. This one is strange.

    :-1: error: framework not found -Wl

For some reason my clang++ is not behaving as expected. Trying to figure this out still. Here is the full output:

    make VERBOSE=1
    ( test -e Makefile.QGCLocationPlugin || /usr/local/Qt5.5.1/5.5/clang_64/bin/qmake /Users/dagoodma/Documents/ASL/dubins_area_coverage/qgroundcontrol/QGCLocationPlugin.pro -spec macx-clang CONFIG+=x86_64 -o Makefile.QGCLocationPlugin ) && make -f Makefile.QGCLocationPlugin
    make[1]: Entering directory '/Users/dagoodma/Documents/ASL/dubins_area_coverage/qgroundcontrol/build-release'
    make[1]: Nothing to be done for 'first'.
    make[1]: Leaving directory '/Users/dagoodma/Documents/ASL/dubins_area_coverage/qgroundcontrol/build-release'
    ( test -e Makefile.QGCApplication || /usr/local/Qt5.5.1/5.5/clang_64/bin/qmake /Users/dagoodma/Documents/ASL/dubins_area_coverage/qgroundcontrol/QGCApplication.pro -spec macx-clang CONFIG+=x86_64 -o Makefile.QGCApplication ) && make -f Makefile.QGCApplication
    make[1]: Entering directory '/Users/dagoodma/Documents/ASL/dubins_area_coverage/qgroundcontrol/build-release'
    echo "extern \"C\" { const char *git_version() { return \"v2.7.1-478-g92cc42a\"; } }" > git_version.cpp && clang++ -c -g git_version.cpp
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -headerpad_max_install_names -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.8.sdk -mmacosx-version-min=10.6 -F/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/Library/Frameworks -weak_framework -Wl,-rpath,/usr/local/Qt5.5.1/5.5/clang_64/lib -o release/qgroundcontrol.app/Contents/MacOS/qgroundcontrol build-release/obj/qextserialport.o build-release/obj/qextserialenumerator.o build-release/obj/qextserialport_unix.o build-release/obj/qextserialenumerator_osx.o build-release/obj/qwt_abstract_legend.o build-release/obj/qwt_abstract_scale.o build-release/obj/qwt_abstract_scale_draw.o build-release/obj/qwt_abstract_slider.o build-release/obj/qwt_analog_clock.o build-release/obj/qwt_arrow_button.o build-release/obj/qwt_clipper.o build-release/obj/qwt_color_map.o build-release/obj/qwt_compass.o build-release/obj/qwt_compass_rose.o build-release/obj/qwt_counter.o build-release/obj/qwt_curve_fitter.o build-release/obj/qwt_dial.o build-release/obj/qwt_dial_needle.o build-release/obj/qwt_dyngrid_layout.o build-release/obj/qwt_event_pattern.o build-release/obj/qwt_graphic.o build-release/obj/qwt_interval.o build-release/obj/qwt_knob.o build-release/obj/qwt_legend.o build-release/obj/qwt_legend_data.o build-release/obj/qwt_legend_label.o build-release/obj/qwt_magnifier.o build-release/obj/qwt_math.o build-release/obj/qwt_null_paintdevice.o build-release/obj/qwt_painter.o build-release/obj/qwt_painter_command.o build-release/obj/qwt_panner.o build-release/obj/qwt_picker.o build-release/obj/qwt_picker_machine.o build-release/obj/qwt_pixel_matrix.o build-release/obj/qwt_plot.o build-release/obj/qwt_plot_axis.o build-release/obj/qwt_plot_canvas.o build-release/obj/qwt_plot_curve.o build-release/obj/qwt_plot_dict.o build-release/obj/qwt_plot_grid.o build-release/obj/qwt_plot_item.o build-release/obj/qwt_plot_layout.o build-release/obj/qwt_plot_magnifier.o build-release/obj/qwt_plot_marker.o build-release/obj/qwt_plot_panner.o build-release/obj/qwt_plot_picker.o build-release/obj/qwt_plot_rasteritem.o build-release/obj/qwt_plot_scaleitem.o build-release/obj/qwt_plot_seriesitem.o build-release/obj/qwt_plot_spectrogram.o build-release/obj/qwt_plot_svgitem.o build-release/obj/qwt_plot_xml.o build-release/obj/qwt_plot_zoomer.o build-release/obj/qwt_point_mapper.o build-release/obj/qwt_point_data.o build-release/obj/qwt_raster_data.o build-release/obj/qwt_round_scale_draw.o build-release/obj/qwt_scale_div.o build-release/obj/qwt_scale_draw.o build-release/obj/qwt_scale_engine.o build-release/obj/qwt_scale_map.o build-release/obj/qwt_scale_widget.o build-release/obj/qwt_series_data.o build-release/obj/qwt_slider.o build-release/obj/qwt_spline.o build-release/obj/qwt_symbol.o build-release/obj/qwt_text.o build-release/obj/qwt_text_engine.o build-release/obj/qwt_text_label.o build-release/obj/qwt_thermo.o build-release/obj/qwt_transform.o build-release/obj/qwt_wheel.o build-release/obj/qwt_widget_overlay.o build-release/obj/QGCAudioWorker.o build-release/obj/CmdLineOptParser.o build-release/obj/LinkConfiguration.o build-release/obj/LinkManager.o build-release/obj/LogReplayLink.o build-release/obj/MAVLinkProtocol.o build-release/obj/MockLink.o build-release/obj/MockLinkFileServer.o build-release/obj/MockLinkMissionItemHandler.o build-release/obj/TCPLink.o build-release/obj/UDPLink.o build-release/obj/FlightDisplayWidget.o build-release/obj/FlightDisplayViewController.o build-release/obj/FlightMapSettings.o build-release/obj/GAudioOutput.o build-release/obj/HomePositionManager.o build-release/obj/Joystick.o build-release/obj/JoystickManager.o build-release/obj/LogCompressor.o build-release/obj/main.o build-release/obj/MissionEditorController.o build-release/obj/MissionManager.o build-release/obj/QGC.o build-release/obj/QGCApplication.o build-release/obj/QGCComboBox.o build-release/obj/QGCDockWidget.o build-release/obj/QGCFileDialog.o build-release/obj/QGCLoggingCategory.o build-release/obj/QGCPalette.o build-release/obj/QGCQmlWidgetHolder.o build-release/obj/QGCQuickWidget.o build-relea [... cut ...]
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    Makefile.QGCApplication:1421: recipe for target 'release/qgroundcontrol.app/Contents/MacOS/qgroundcontrol' failed
    make[1]: *** [release/qgroundcontrol.app/Contents/MacOS/qgroundcontrol] Error 1
    make[1]: Leaving directory '/Users/dagoodma/Documents/ASL/dubins_area_coverage/qgroundcontrol/build-release'
    Makefile:70: recipe for target 'sub-QGCApplication-pro-make_first-ordered' failed
    make: *** [sub-QGCApplication-pro-make_first-ordered] Error 2

dagoodma, 2015-10-18:
I compiled llvm 3.6.2 from source to upgrade my clang compiler, but I am still seeing the same issue when running make:

    ld: framework not found -Wl
    clang: error: linker command failed with exit code 1 (use -v to see invocation)

Here is my clang version:

    clang version 3.8.0
    Target: x86_64-apple-darwin11.4.2

DonLakeFlyer, 2015-10-18:
I'm going to close this. we'll try to help, but you are fundamentally doing something we don't support

closed 2015-10-18 by DonLakeFlyer