mavlink/qgroundcontrol #2030: Waypoint dragging visuals

LorenzMeier, 2015-10-18:
Waypoint dragging works well, but the visual needs to indicate that its selected (e.g. by growing it 10x)
and it would be good if a transparent line moves with it.

DonLakeFlyer, 2015-10-18:
Not going to be able to move a line with the drag without a ton of work.

dogmaphobic, 2015-10-18:
> Not going to be able to move a line with the drag without a ton of work.

I had that working in the original code. I will dig it up...

DonLakeFlyer, 2015-10-18:
You know I don't know what the hell I was doing but I somehow missed all that code. Most likely because I developed the Mission Editor in a separate vacuum (don't tell Lorenz!). And in the frenzy to tear out the old map stuff I deleted right over it. My screw up.

The reason why it's more complicated now is because as you know you can't just move MapPolyLine coordinates. You basically need to recalc the lines yourself anytime anything changes. The line recalc code doesn't just put a line in between all the mission items. That recalc gets complicated because a line is only connected to the home position when the next item is a takeoff item. It is also complicated by the fact that not all mission items have a coordinate associated with them. Lines don't go through those items. So determining what one/two/none lines move around when you drag an item is a little tricky.

But in fact as I type all this I'm realizing it's actually not too terrible. Just a small PITA. In fact it could be made even easier if you aren't ghosting a new set of lines but just live recalcing the full line set. I'm not sure if that will work from a redraw standpoint. But it's worth a try.

dogmaphobic, 2015-10-19:
Yep, I never got around to it today. What I remember was that I would replace the entire coordinate array each time (as you can't change a single point). That seemed to be completely fine time wise. I didn't notice any tearing or degradation in the rendering. Even with dozens of points.

DonLakeFlyer, 2015-10-19:
I'll try live drag, with recalc on every position change. We'll see whaat happens.

DonLakeFlyer, 2015-10-24:
Closing this. We may or may not add line tracking during drag. But it's not critical

closed 2015-10-24 by DonLakeFlyer