mavlink/qgroundcontrol #2031: Pinch gesture halfway working

LorenzMeier, 2015-10-18:
On Qt 5.5.1 the pinch gesture works on the edit view reasonably well, but doesn't work on the fly view.

DonLakeFlyer, 2015-10-22:
Even on Qt 5.5.1 FlightDisplayView needs a bogus MouseArea workaround. I pulled the one out of FlightMap because I though it would all work on 5.5.1 (which of course was wrong) and I was getting strange results with all my MouseArea's in the Editor.

dogmaphobic, 2015-10-23:
I will bet it's got something to do with QtQuickWidget. They've broken a whole lot of things in there in 5.5.x. Pinch zoom using the location samples now work just fine, without the hack. Those samples are regular QtQuick apps, not QtQuickWidget.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier