mavlink/qgroundcontrol #2033: Decimal places with 5.5.1

LorenzMeier, 2015-10-18:
The good old bug with decimal places after the comma. Wasn't this supposed to be fixed?

DonLakeFlyer, 2015-10-18:
I'm going to need to fix that on our side of things.

DonLakeFlyer, 2015-10-22:
@LorenzMeier What I really need is a precision value in the parameter meta data for floats. Otherwise I have no idea that:

LorenzMeier, 2015-10-23:
You already have min and max, which encodes the significant decimal places. Max - min / 10'000 steps or so should give you the number of digits.

DonLakeFlyer, 2015-10-23:
That doesn't work, Two reasons:
- Most params are missing min/max
- Example with BAT_V_CHARGED (which has no min/max set). Let make some up: max=4.2, min=3.8. max-min/10000=4000. No sure how that tells me how many digits past the decimal point I want. For this example the answer I want is 1. No reason to show any more than 1 digit past decimal in ui like 4.2 and 3.8.

LorenzMeier, 2015-10-23:
I see the point. Well, then let's add the meta info and let it default to 3 digits if none is available.

DonLakeFlyer, 2015-10-23:
3 sound good. I'll add the support on the QGC side in the FactMetaData. Can you get someone to do the firmware side work. Parser update, plus add the meta data.

LorenzMeier, 2015-10-23:
I found "someone" and named the tag "decimal". It will go into all apps over time.
https://github.com/PX4/Firmware/commits/master

DonLakeFlyer, 2015-10-23:
Perfect thanks.

closed 2015-10-25 by DonLakeFlyer