mavlink/qgroundcontrol #2053: Can't Run or Build

zshn25, 2015-10-20:
It was working fine, But yesterday, I got this error from QT Creator while Running QGCS
```
:-1: error: dependent 'C:\Users\Public\Desktop\qgroundcontrol\build-QGCApplication-Desktop_Qt_5_4_0_MSVC2013_OpenGL_32bit-Debug\build-debug\obj\QGeoServiceProviderFactoryQGC.lib' does not exist.
```

Then I tried to Build the sln file from Visual Studio, I got these errors

```
Warning	1	warning MSB8028: The intermediate directory (build-release\obj\) contains files shared from another project (QGeoServiceProviderFactoryQGC.vcxproj).  This can lead to incorrect clean and rebuild behavior.	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\Microsoft.CppBuild.targets	388	5	qgroundcontrol
Warning	2	warning D9025: overriding '/Zc:strictStrings' with '/Zc:strictStrings-'	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\cl	qgroundcontrol
Warning	3	warning D9025: overriding '/Zc:strictStrings' with '/Zc:strictStrings-'	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\cl	qgroundcontrol
Warning	4	warning D9025: overriding '/Zc:strictStrings' with '/Zc:strictStrings-'	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\cl	qgroundcontrol
Error	5	error C2146: syntax error : missing ';' before identifier 'coordinate' (src\MissionItemTest.cc)	c:\users\public\desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.h	49	1	qgroundcontrol
Error	6	error C2146: syntax error : missing ';' before identifier 'coordinate' (build-release\moc\moc_MissionItemTest.cpp)	c:\users\public\desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.h	49	1	qgroundcontrol
Error	7	error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (src\MissionItemTest.cc)	c:\users\public\desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.h	49	1	qgroundcontrol
Error	8	error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (build-release\moc\moc_MissionItemTest.cpp)	c:\users\public\desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.h	49	1	qgroundcontrol
Error	9	error C3861: 'QGeoCoordinate': identifier not found	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	29	1	qgroundcontrol
Error	10	error C3861: 'QGeoCoordinate': identifier not found	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	30	1	qgroundcontrol
Error	11	error C3861: 'QGeoCoordinate': identifier not found	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	31	1	qgroundcontrol
Error	12	error C3861: 'QGeoCoordinate': identifier not found	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	32	1	qgroundcontrol
Error	13	error C3861: 'QGeoCoordinate': identifier not found	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	33	1	qgroundcontrol
Error	14	error C3861: 'QGeoCoordinate': identifier not found	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	34	1	qgroundcontrol
Error	15	error C3861: 'QGeoCoordinate': identifier not found	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	35	1	qgroundcontrol
Error	16	error C3861: 'QGeoCoordinate': identifier not found	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	36	1	qgroundcontrol
Warning	17	warning C4305: 'initializing' : truncation from 'double' to 'bool'	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	37	1	qgroundcontrol
Error	18	error C2078: too many initializers	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	37	1	qgroundcontrol
Error	19	error C2065: 'MissionItem' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	122	1	qgroundcontrol
Error	20	error C2065: 'item' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	122	1	qgroundcontrol
Error	21	error C2061: syntax error : identifier 'MissionItem'	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	122	1	qgroundcontrol
Error	22	error C2065: 'item' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	137	1	qgroundcontrol
Error	23	error C2227: left of '->save' must point to class/struct/union/generic type	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	137	1	qgroundcontrol
Error	24	error C2065: 'item' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	142	1	qgroundcontrol
Error	25	error C2227: left of '->textFieldFacts' must point to class/struct/union/generic type	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	142	1	qgroundcontrol
Error	26	error C2227: left of '->count' must point to class/struct/union/generic type	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	142	1	qgroundcontrol
Error	27	error C2065: 'Fact' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	143	1	qgroundcontrol
Error	28	error C2065: 'fact' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	143	1	qgroundcontrol
Error	29	error C2059: syntax error : '>'	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	143	1	qgroundcontrol
Error	30	error C2065: 'item' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	143	1	qgroundcontrol
Error	31	error C2227: left of '->textFieldFacts' must point to class/struct/union/generic type	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	143	1	qgroundcontrol
Error	32	error C2227: left of '->get' must point to class/struct/union/generic type	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	143	1	qgroundcontrol
Error	33	error C2065: 'fact' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	149	1	qgroundcontrol
Error	34	error C2227: left of '->name' must point to class/struct/union/generic type	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	149	1	qgroundcontrol
Error	35	error C2065: 'fact' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	151	1	qgroundcontrol
Error	36	error C2227: left of '->value' must point to class/struct/union/generic type	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	151	1	qgroundcontrol
Error	37	error C2228: left of '.toDouble' must have class/struct/union	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	151	1	qgroundcontrol
Error	38	error C2065: 'MissionItem' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	163	1	qgroundcontrol
Error	39	error C2065: 'loadedItem' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	163	1	qgroundcontrol
Error	40	error C2061: syntax error : identifier 'MissionItem'	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	163	1	qgroundcontrol
Error	41	error C2065: 'loadedItem' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	165	1	qgroundcontrol
Error	42	error C2227: left of '->load' must point to class/struct/union/generic type	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	165	1	qgroundcontrol
Error	43	error C2065: 'loadedItem' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	167	1	qgroundcontrol
Error	44	error C2227: left of '->coordinate' must point to class/struct/union/generic type	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	167	1	qgroundcontrol
Error	45	error C2228: left of '.latitude' must have class/struct/union	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	167	1	qgroundcontrol
Error	46	error C2065: 'item' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	167	1	qgroundcontrol
Error	47	error C2661: 'QTest::qCompare' : no overloaded function takes 5 arguments	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	167	1	qgroundcontrol
[... cut ...]
Error	91	error C2541: 'delete' : cannot delete objects that are not pointers	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	179	1	qgroundcontrol
Error	92	error C2065: 'loadedItem' : undeclared identifier	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	180	1	qgroundcontrol
Error	93	error C2541: 'delete' : cannot delete objects that are not pointers	C:\Users\Public\Desktop\qgroundcontrol\qgroundcontrol\src\MissionItemTest.cc	180	1	qgroundcontrol
```

Please Help!!

NaterGator, 2015-10-20:
When you imported the project in Qt Creator did you import using qgroundcontrol.pro or QGCApplication.pro? You want to import using qgroundcontrol.pro to ensure that the dependent QGCLocationPlugin is built before the QGroundControl application.

zshn25, 2015-10-23:
I imported qgroundcontrol.pro itself.

DonLakeFlyer, 2015-10-23:
I'd clean everything out. And start over using the new readme intructions.

zshn25, 2015-10-23:
Tried that too. Actually, the Debug build is not showing any errors but
Release build has these errors

DonLakeFlyer, 2015-10-23:
There are lots of magic files that QtCreator keeps around. You should find where those files are kept on Windows (AppData somewhere). On OS X they are ~/.config/Qt Project/*. Delete those as well to clean your installation. Sometimes thing get out of whack with those. I've had that problem more in Windows than any other OS.

NaterGator, 2015-10-23:
If it is only happening in release I would be suspicious of having shadow build enabled.

I remember having some issues with shadow build support enabled in Windows.

zshn25, 2015-10-24:
So, how do I disable this Shadow Build or remove magic files from Windows.

DonLakeFlyer, 2015-10-24:
Shadow build settings is on the projects page of Qt Creator. Like I said before Qt Creator settings file are in one of the Windows AppData directories. Directory name should be Qt Project I think.

DonLakeFlyer, 2015-10-24:
I just built using a Shadow Build that I had previously built debug it without problems. Release spits out some warnings on strictstrings but it builds an executable fine. The Qt Creator settings are in AppData/Roaming/QtProject. We can continue the discussion here to try to help. But the problem is specific to your machine.

closed 2015-10-24 by DonLakeFlyer

zshn25, 2016-05-04:
Hey mavlink/qgroundcontr,

Discover high quality jobs and connect with recruiters on LinkedIn.

- Zeeshan

Accept invite: https://www.linkedin.com/e/v2?e=0-ins5rflo-fp&t=ssuw&ek=invite_guest&li=29&m=invite_guest&ts=india_a&sharedKey=KawmPeUk&invitationId=6133430001304948736

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=0-ins5rflo-fp&t=lun&midToken=AQFmrFVhGEd36Q&ek=invite_guest&loid=AQF4uEMnxA9zewAAAVR5ShCSAp02341F4_xaPlLx0pwxAZZlkqg83Q&eid=0-ins5rflo-fp

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=0-ins5rflo-fp&a=customerServiceUrl&ek=invite_guest

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

zshn25, 2016-05-04:
Hey mavlink/qgroundcontr,

Discover high quality jobs and connect with recruiters on LinkedIn.

- Zeeshan

Accept invite: https://www.linkedin.com/e/v2?e=0-ins5rf00-wj&t=ssuw&ek=invite_guest&li=29&m=invite_guest&ts=india_a&sharedKey=bfWyWINQ&invitationId=6133430001976045568

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=0-ins5rf00-wj&t=lun&midToken=AQFmrFVhGEd36Q&ek=invite_guest&loid=AQEK6BfOYVLWuwAAAVR5TCrJswPxlaX-ogtw1CJYbxI1zuA1jZQdDw&eid=0-ins5rf00-wj

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=0-ins5rf00-wj&a=customerServiceUrl&ek=invite_guest

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

zshn25, 2016-05-10:
Zeeshan Khan would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=3qlwix-io16a82b-4x&a=preRegInvite&ek=first_guest_reminder_01&li=14&m=hero&ts=accept_text&sharedKey=KawmPeUk&invitationID=6133430001304948736

View Zeeshan Khan&#39;s profile: https://www.linkedin.com/e/v2?e=3qlwix-io16a82b-4x&a=preRegInvite&ek=first_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=KawmPeUk&invitationID=6133430001304948736

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=3qlwix-io16a82b-4x&t=lun&midToken=AQFmrFVhGEd36Q&ek=first_guest_reminder_01&li=16&m=unsub&ts=HTML&eid=3qlwix-io16a82b-4x&loid=AQE0sO9sSX5WGgAAAVSawfRgorIO0kuKRPMsSTGaexYrCBC5bkTvC9p1YmZJt1RAnY4XntHxfBBJu_YQ04D36PM1Pn0PL19ojTzKiqIsOFMcNHd9-ogGIOaFtPSPgyQRqHS2Ydi2-MHkgAmSAVfyo0lnRrNtvUViG4FeqbVafPBWBtXoWNWyUSGiAWtaFddtgJPv

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=3qlwix-io16a82b-4x&a=customerServiceUrl&ek=first_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

zshn25, 2016-05-10:
Zeeshan Khan would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=vgtvzb-io16bis7-jh&a=preRegInvite&ek=first_guest_reminder_01&li=14&m=hero&ts=accept_text&sharedKey=bfWyWINQ&invitationID=6133430001976045568

View Zeeshan Khan&#39;s profile: https://www.linkedin.com/e/v2?e=vgtvzb-io16bis7-jh&a=preRegInvite&ek=first_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=bfWyWINQ&invitationID=6133430001976045568

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=vgtvzb-io16bis7-jh&t=lun&midToken=AQFmrFVhGEd36Q&ek=first_guest_reminder_01&li=16&m=unsub&ts=HTML&eid=vgtvzb-io16bis7-jh&loid=AQFrt06_D8PhywAAAVSawp_rAuYykUIPUVcmJBE15vfYhUF7cQx0UPcocrrbVEG4NmT6h51hgsCEnVpLxmmLJAFrB-0bKs4ZP-MDpuSuAC-NcTmSBWGJiNPLRkg8mtDgeXpq5RuXPOVf-tm5tGkzftIidUp__TaXRh3LXq4nVNR3qFrC4WMoh7Bcuazlo4mOERGI

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=vgtvzb-io16bis7-jh&a=customerServiceUrl&ek=first_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

zshn25, 2016-05-17:
Zeeshan Khan would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=3qlwix-iob6bt2u-ma&t=ssuw&ek=second_guest_reminder_01&li=13&m=hero&ts=accept_text&sharedKey=KawmPeUk&invitationId=6133430001304948736

View Zeeshan Khan&#39;s profile: https://www.linkedin.com/e/v2?e=3qlwix-iob6bt2u-ma&t=ssuw&ek=second_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=KawmPeUk&invitationId=6133430001304948736

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=3qlwix-iob6bt2u-ma&t=lun&midToken=AQFmrFVhGEd36Q&ek=second_guest_reminder_01&li=15&m=unsub&ts=HTML&eid=3qlwix-iob6bt2u-ma&loid=AQGihJyomoiTOAAAAVS-zsBDfrVg8mWQA9QaEYWx3cB3Wt1r9QCzGFY5mqMTJszqnBCUPCWPOCWFU2Twg1XpNlRYnyV6qYH8RVRvhyNTmWaSPrR3c-VIMLBKv7yctX38rcxFHq4suHPns363GSc3e4h8XtvFEeEOWHR0zX6alwGdnvoS4PmafNEHXYjHrzksiHxG

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=3qlwix-iob6bt2u-ma&a=customerServiceUrl&ek=second_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

zshn25, 2016-05-17:
Zeeshan Khan would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=vgtvzb-iob6c6wo-7x&t=ssuw&ek=second_guest_reminder_01&li=13&m=hero&ts=accept_text&sharedKey=bfWyWINQ&invitationId=6133430001976045568

View Zeeshan Khan&#39;s profile: https://www.linkedin.com/e/v2?e=vgtvzb-iob6c6wo-7x&t=ssuw&ek=second_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=bfWyWINQ&invitationId=6133430001976045568

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=vgtvzb-iob6c6wo-7x&t=lun&midToken=AQFmrFVhGEd36Q&ek=second_guest_reminder_01&li=15&m=unsub&ts=HTML&eid=vgtvzb-iob6c6wo-7x&loid=AQHp8r0VCOxLKwAAAVS-zwFJISS6kGeVn3eCkxVlhQ-U5PbZ7UvVY0zS2EO70QtR1m53icYqY34cXi0PYtJ9HLyDLja5_5E1eHSwNVe0bx_LiNdVSoL7iQGwDJ4AB1DE6X-k7pJMUHtvWqWiVAFGdIIfk-cpy0w01XUG9rd4B70tGNug3vIydv3dspYVMlz8CR6t

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=vgtvzb-iob6c6wo-7x&a=customerServiceUrl&ek=second_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.