mavlink/qgroundcontrol #2054: Build instructions are too long and too complex

LorenzMeier, 2015-10-20:
I've learned a lot about how to write documentation that people will actually read and going back to our README there seems room for improvement:

  * It's generally way too much text
  * Much of the complexity is brought in by not defaulting to Qt Creator for all platforms. By doing that and providing 2-3 screenshots everything could be covered more intuitively
  * Pro instructions should be clearly separated
  * Now that 5.5.1 is out we might be able to do away with some of the magic

I have a lot of students failing to build QGC on Ubuntu 14.04. Because current stable is hopelessly outdated and has the UDP connecting bug they can't use binaries.

dagar, 2015-10-20:
I completely agree. I wasn't sure how to handle updating the ubuntu instructions for 5.5.1
https://github.com/mavlink/qgroundcontrol/pull/2050

The detailed instructions tend to break due to things out of our control like the Qt 5.4.1/5.4.2 ppa for ubuntu. More advanced don't need instructions for installing qt packages, using qmake on the command line, etc. Lengthy detailed instructions have useful information, but most people just skip walls of text.

I think we really only need the following
 1. os specific dependencies
    - apt-get install libsdl1.2-dev, yum install SDL, etc
    - windows compiler?
    - mac os?
 2. install Qt 5.5.1 for your platform http://www.qt.io/download/ or use packages if you prefer
 3. launch qtcreator and open qgroundcontrol/qgroundcontrol.pro

DonLakeFlyer, 2015-10-20:
I agree. By just doing 2 and 3 everything ends up the same for all OS.

NaterGator, 2015-10-20:
Regarding 2, point them directly to http://www.qt.io/download-open-source/ so they don't have to answer obtuse licensing questions that encourage them into a commercial Qt license. 

If we default to QtCreator it might be worthwhile to invert the [jombuild](https://github.com/mavlink/qgroundcontrol/blob/a39090fbbe93d8a3c0b72eb7574694e4da26b498/QGCCommon.pri#L169-L175) behavior so we default to efficient parallel compilation in QtCreator and display a message explaining config options for people off the beaten path using Visual Studio.

DonLakeFlyer, 2015-10-22:
Fixed

closed 2015-10-22 by DonLakeFlyer