mavlink/qgroundcontrol #2060: No rc messages generated outside of Joystick tab

tumbili, 2015-10-21:
If I comment this line here https://github.com/mavlink/qgroundcontrol/blob/master/src/Joystick/Joystick.cc#L530 everything works fine.
Who activates the polling for Joystick messages normally?

LorenzMeier, 2015-10-21:
Does it still work if you never ever enter the tab (e.g. boot QGC fresh then just try).

tumbili, 2015-10-21:
Potentially fixed here #2061

DonLakeFlyer, 2015-10-22:
The enabled checkbox on the joystick config screen turns on polling that will continue after you leave the config screen. Keep in mind that joystick config uses COM_RC_IN_MODE=1 by default which does not emulate RC.

DonLakeFlyer, 2015-10-25:
Can you give me some details on exactly what is going wrong here (from an end user standpoint) and how to repro?

LorenzMeier, 2015-10-25:
@DonLakeFlyer This call seems to be only executed when you pass through the joystick config widget:
```C
Vehicle::setJoystickEnabled(_joystickEnabled);
```

I couldn't find a reference to it outside of the config UI, which is consistent with our testing (joystick only responsive if you visited the config UI once at least). It might have been fixed since, but might be worth checking if you enable the joystick from settings.

tumbili, 2015-10-26:
@DonLakeFlyer I think @devbharat still has  this problem with his Taranis. Basically the rc channels are not emulated any more in QGC when your are inside the Joystick tab. As soon as you click on another tab there is no response to the Joystick.
@devbharat Could you do a clean build and check if you still have this problem?
I'm finding this strange because I used to have this problem but not any more with latest master.
I can verify that it's working with the Taranis today in the lab.

devbharat, 2015-10-26:
Will do

tumbili, 2015-10-30:
@DonLakeFlyer @LorenzMeier I still have this issue with latest QGC. As soon as I leave the Joystick tab the rc channels are not created anymore. This makes it impossible to calibrate the radio.

@devbharat If you want to try the new QGC, comment this line here as a quick hack if you have the same issue as I do:
https://github.com/mavlink/qgroundcontrol/blob/master/src/Joystick/Joystick.cc#L540

DonLakeFlyer, 2015-10-30:
> Keep in mind that joystick config uses COM_RC_IN_MODE=1 by default which does not emulate RC.

If you want to emulate RC (where it actually sends RC_CHANNELS mavlink messages) you need to manually set COM_RC_IN_MODE=2 yourself after you calibrate joystick. Curious as to why you would want to use this mode instead of the default?

LorenzMeier, 2015-10-30:
Don: It's not sending any joystick data out. None. Its not a mode thing.

LorenzMeier, 2015-10-30:
PR to fix:
https://github.com/mavlink/qgroundcontrol/pull/2164

tumbili, 2015-10-30:
Thanks, fixed, closing.

closed 2015-10-30 by tumbili