mavlink/qgroundcontrol #2062: SDL not found

LorenzMeier, 2015-10-21:
dyld: Library not loaded: @rpath/SDL.framework/Versions/A/SDL
  Referenced from: /Users/lorenzmeier/src/build-qgroundcontrol-Desktop_Qt_5_4_1_clang_64bit-Release/release/qgroundcontrol.app/Contents/MacOS/qgroundcontrol
  Reason: image not found
The program has unexpectedly finished.

LorenzMeier, 2015-10-21:
This happens after a spurious "cp", command not found message on the compile output. I also had a "rm" not found on a 2nd machine yesterday.

LorenzMeier, 2015-10-21:
Here is the compile fail:
```
rm -f libQGeoServiceProviderFactoryQGC.a
make[1]: rm: No such file or directory
make[1]: [build-release/obj/libQGeoServiceProviderFactoryQGC.a] Error 1 (ignored)
make[1]: rm: No such file or directory
make[1]: [build-release/obj/libQGeoServiceProviderFactoryQGC.a] Error 1 (ignored)
make[1]: mv: No such file or directory
make[1]: [build-release/obj/libQGeoServiceProviderFactoryQGC.a] Error 1 (ignored)
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ar cq libQGeoServiceProviderFactoryQGC.a build-release/obj/qgeoserviceproviderpluginqgc.o build-release/obj/qgeotiledmappingmanagerengineqgc.o build-release/obj/qgeotilefetcherqgc.o build-release/obj/qgeomapreplyqgc.o build-release/obj/qgeocodingmanagerengineqgc.o build-release/obj/qgeocodereplyqgc.o build-release/obj/OpenPilotMaps.o build-release/obj/moc_qgeoserviceproviderpluginqgc.o build-release/obj/moc_qgeotiledmappingmanagerengineqgc.o build-release/obj/moc_qgeotilefetcherqgc.o build-release/obj/moc_qgeomapreplyqgc.o build-release/obj/moc_qgeocodingmanagerengineqgc.o build-release/obj/moc_qgeocodereplyqgc.o build-release/obj/moc_OpenPilotMaps.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib -s libQGeoServiceProviderFactoryQGC.a
rm -f build-release/obj/libQGeoServiceProviderFactoryQGC.a
mv -f libQGeoServiceProviderFactoryQGC.a build-release/obj/ 
( test -e Makefile.QGCApplication || /Users/user/Qt/5.4/clang_64/bin/qmake /Users/user/src/qgroundcontrol/QGCApplication.pro -spec macx-clang CONFIG+=x86_64 -o Makefile.QGCApplication ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile.QGCApplication 
/bin/sh: mkdir: command not found
/bin/sh: mkdir: command not found
make[1]: *** No rule to make target `/Users/user/src/build-qgroundcontrol-Desktop_Qt_5_4_1_clang_64bit-Release/build-release/obj/libQGeoServiceProviderFactoryQGC.a', needed by `release/qgroundcontrol.app/Contents/MacOS/qgroundcontrol'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [release/qgroundcontrol.app/Contents/Resources/empty.lproj] Error 127
make[1]: *** [release/qgroundcontrol.app/Contents/PkgInfo] Error 127
/bin/sh: mkdir: command not found
make[1]: *** [release/qgroundcontrol.app/Contents/Info.plist] Error 127
/bin/sh: mkdir: command not found
make[1]: *** [release/qgroundcontrol.app/Contents/Resources/macx.icns] Error 127
make: *** [sub-QGCApplication-pro-make_first-ordered] Error 2
10:25:46: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Desktop Qt 5.4.1 clang 64bit)
When executing step "Make"
10:25:46: Elapsed time: 00:20.
```

LorenzMeier, 2015-10-21:
@dogmaphobic This seems to be libQGeoServiceProviderFactoryQGC related. I can't build any more with Qt 5.4 or 5.5.1.

LorenzMeier, 2015-10-21:
I'm de-installed Qt and will re-install Qt 5.5.1 clean. Lets see if that fixes it.

dogmaphobic, 2015-10-21:
Hmm... I have not seen any build issues. I just merged master to make sure, deleted the build directory and run a full build. No issues:
```
08:11:07: The process "/usr/bin/make" exited normally.
08:11:07: Elapsed time: 01:16.
```

dogmaphobic, 2015-10-21:
Just tried Qt 5.4.2 for the heck of it and it had no issues either:

```
08:13:54: Running steps for project qgroundcontrol...
08:13:54: Starting: "/Volumes/Data/local/Qt/5.4/clang_64/bin/qmake" /Users/gus/GitHub/qgroundcontrol/qgroundcontrol.pro -r -spec macx-clang CONFIG+=debug CONFIG+=x86_64 CONFIG+=declarative_debug CONFIG+=qml_debug
Project MESSAGE: Qt version 5.4.2
Project MESSAGE: QGroundControl version v2.7.1-547-g77be5c4
Project MESSAGE: Mac build
<...>
08:15:06: The process "/usr/bin/make" exited normally.
08:15:06: Elapsed time: 01:12.
```

DonLakeFlyer, 2015-10-21:
I've had the shadow build blow up on me in similar strange ways. In order to clean I've had to delete all the build directories as well as clean up the QtCreator cached information stored in ~/.config/QtProject. Those magic files in /config sometimes get screwed up.

LorenzMeier, 2015-10-21:
Fixed by reinstalling Qt.

closed 2015-10-21 by LorenzMeier