mavlink/qgroundcontrol #2075: Ubuntu package dependencies incomplete

LorenzMeier, 2015-10-22:
@dagar I assume the dependencies are imcomplete: We're missing (in theory their runtime equivalents, but I would fall back onto the dev libs for now):

* espeak
* libespeak-dev
* libudev-dev
* libsdl1.2-dev

And probably parts of Qt:
```
./qgroundcontrol: error while loading shared libraries: libQt5QuickWidgets.so.5: cannot open shared object file: No such file or directory
```

LorenzMeier, 2015-10-22:
@DonLakeFlyer FYI

LorenzMeier, 2015-10-22:
PPA is also complaining: https://launchpad.net/~qgroundcontrol/+archive/ubuntu/ppa

dagar, 2015-10-22:
Whoops. I'll separate runtime and development requirements. For running should we just skip the Qt installer (and PATH update that I missed) and throw the libs into the linux "package" (tar.bz2)?

dagar, 2015-10-22:
I think we'll have to ignore the PPA until we can find a source for 5.5.1 packages.

LorenzMeier, 2015-10-22:
The ppa got renamed to 5.4.2 and is still available.

I think we should add an installation test but probably run it only for the branch we create the installers on.

DonLakeFlyer, 2015-10-22:
I'm tossing all the ppa stuff an moving everything to online installer. If folks want to get fancy they are on their one. Working on it now. See the new readme. Almost finished. Verifying that it all works on new windows install, and clean Ubuntu vm.

closed 2015-10-22 by DonLakeFlyer