mavlink/qgroundcontrol #2077: Android Text To Speech needs queueing

dogmaphobic, 2015-10-22:
As it is, calls made to the TTS engine on Android has to be done from the main thread, where we can't *wait* for a current message to end before sending a new one. We need something that would allow messages to be queued. Right now, a new message will interrupt an existing, ongoing one and replace it.

Not sure if this is a blessing or a curse. Try flicking the mode switch on your RC a few times. You will get a barrage of messages that are mostly, irrelevant as only the last one is important.

NaterGator, 2016-04-07:
@dogmaphobic I want to revive this and bring  @DonLakeFlyer / @LorenzMeier into the fold. The current TTS system is just not very useful, it is overwhelmed with noise from the autopilot and in the event of a great deal of messages falls very far behind (to the point where it may be speaking _incorrect_ state information).

I think we should implement a much better queuing system. It should:
- Have the ability to skip reading certain messages 
- Prioritize safety critical information to be spoken first
- Drop messages that relay incorrect state (like a mode change that is no longer the current mode) 
- Allow messages to expire after sitting in the queue for too long
- Speak descriptive phrases instead of coded / enum values (where possible)

Just generally the signal to noise ratio can drop; the user will tune out the TTS system if it never has anything useful or correct to say.

dogmaphobic, 2016-04-07:
Been there quite a bit :) I usually just turn off TTS as it's mostly just annoying.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier