mavlink/qgroundcontrol #2089: Mag calibration does not find additional mag

tumbili, 2015-10-23:


DonLakeFlyer, 2015-10-23:
Are you talking about the rotation settings showing up? Did you check the parameters after you calibrated to verify they were set correctly? In order for rotation to show up this must be true:

```
    property bool showCompass0Rot: cal_mag0_id.value > 0 && cal_mag0_rot.value >= 0
    property bool showCompass1Rot: cal_mag1_id.value > 0 && cal_mag1_rot.value >= 0
    property bool showCompass2Rot: cal_mag2_id.value > 0 && cal_mag2_rot.value >= 0
```

tumbili, 2015-10-23:
@LorenzMeier Could you explain what the problem is?

LorenzMeier, 2015-10-23:
@DonLakeFlyer Yes, it didn't show. I need to repro again.

DonLakeFlyer, 2015-10-23:
If those params don't came back from firmware they won't show. After calibration completes, QGC does a full param refresh, which should pull the new values across and in turn show the appropriate controls.

LorenzMeier, 2015-10-23:
Could be a ripple of the param hashing, hmm.

DonLakeFlyer, 2015-10-23:
@NaterGator When is the hash on the QGC side updated?

DonLakeFlyer, 2015-12-03:
Still a problem? Need exact report steps.

NaterGator, 2015-12-03:
@DonLakeFlyer sorry I just saw the comment from Oct 23. The hash is recomputed each time the magic parameter is requested.

DonLakeFlyer, 2015-12-03:
@tumbli ?

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier