mavlink/qgroundcontrol #2091: Verify support for hash-checked parameter cache for APM

NaterGator, 2015-10-23:
Assuming the APM guys are amendable to the idea of adding the _HASH_CHECK magic param [here](https://github.com/diydrones/ardupilot/issues/3070) we should validate the parameter quick-load support [implemented](https://github.com/mavlink/qgroundcontrol/pull/1965) for PX4 also works for APM devices.

DonLakeFlyer, 2015-10-23:
@NaterGator Could you update MockLink to support hash check with a configuration variable? Then we can write a unit test for it as well.

NaterGator, 2015-10-23:
Sure, how do the following test cases sound:
  1.  MockLink ignores _HASH_CHECK request
  2.  MockLink generates _HASH_CHECK value correctly
  3.  MockLink intentionally sabotages _HASH_CHECK value

Case 1 should trigger the timer, case 2 should trigger the cache load, and case 3 should trigger the cache miss and fire PARAM_REQUEST_LIST before the timeout timer fires?

DonLakeFlyer, 2015-10-23:
Yup. If you need help with things like MultiSignalSpy and so forth let me know. Thanks Nate. The new MissionManagerTest has an example of a PX4 plus APM MockLink setup test.

DonLakeFlyer, 2015-10-23:
Actually you don't need to make a PX4 and and APM variant. You test cases cover all the bases.

DonLakeFlyer, 2015-10-24:
While you're in there could you add a --disable-param-hash command line option. Useful to test any strange hash related issues.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier