mavlink/qgroundcontrol #2134: Unsuported Linux toolchain, only GCC 32- or 64-bit is supported

killerlayup, 2015-10-28:
when I build qgroundcontrol in ubuntu 12.04 32bit, QT 5.5.1 version,it comes out an issue:

Project MESSAGE: Qt version 5.5.1
Project ERROR: Unsuported Linux toolchain, only GCC 32- or 64-bit is supported
14:07:09: The process "/home/joel/Qt5.5.1/5.5/gcc/bin/qmake" exited with code 3.
Error while building/deploying project qgroundcontrol (kit: Desktop Qt 5.5.1 GCC 32bit)
When executing step "qmake"

could someone figure this issue?
thanks

killerlayup, 2015-10-28:
can someone help me pls?

LorenzMeier, 2015-10-28:
This is a Qt issue, not a QGroundControl issue. In general 12.04 is most likely just too old to be supported by a modern Qt version. If you use Ubuntu LTS you should not expect support for new functionality, just fixes. If you want new things like Qt5 use a more recent Ubuntu version.

closed 2015-10-28 by LorenzMeier

killerlayup, 2015-10-28:
@LorenzMeier 
thanks for your reply. I have one more question. Can simulation environment HIL in http://pixhawk.org/dev/ros/hil be installed in ARM like odroid? it seemed that jMAVSim can't be install in ARM..

LorenzMeier, 2015-10-28:
This is a Java question. If Java runs on an Odroid then yes.