mavlink/qgroundcontrol #2148: Crash of qGroundControl owing to the console.

AlexisTM, 2015-10-29:
Dear,

When we select something in the console (the log with errors and informations), then "grab" it to qGroundControl window, qGroundControl crash.

DonLakeFlyer, 2015-10-29:
Sorry but I don't understand what you are saying is the problem. QGroundControl has not console build into it. Can you provide very specific details as to what you are doing and/or screen shots.

AlexisTM, 2015-10-30:
Indeed, I do not know how to name it. I did a screenshot.

OS : Windows 10

Reproduce it : 
* Select the text (like in the image)
* Left click and hold on the text (drag)
* Drop the text anywhere : Crash

[image]

DonLakeFlyer, 2015-11-23:
This dropdown has been replaced.

closed 2015-11-23 by DonLakeFlyer