mavlink/qgroundcontrol #2155: Plot euler angles for attitude setpoint

tumbili, 2015-10-29:
The attitude setpoint is just available in the form of a quaternion which is not very useful for tuning.

dagar, 2015-10-29:
Would it make more sense to add them to ATTITUDE_TARGET?

tumbili, 2015-10-29:
We also thought about that. But then all the information is already given by the quaternion. Is there reason why you would prefer adding them to ATTITUDE_TARGET?

dagar, 2015-10-29:
It just seems slightly cleaner to not have special case plotting code, especially considering they're just converted before being sent over mavlink.
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L2007

LorenzMeier, 2015-10-29:
Its not really, because it puts the gimbal lock into that part of transmission. We will be adding a 3D model of the craft soon in QGC to understand its attitude while its in flight, and for VTOL that would lead to the associated problems.

The same is true if you try to draw a HUD at 90 degrees pitch.

LorenzMeier, 2015-10-29:
Will fix this in Firmware:
https://github.com/PX4/Firmware/issues/3098

closed 2015-12-26 by LorenzMeier