mavlink/qgroundcontrol #2162: QGC loads waypoints from vehicle on boot, but doesn't display them

LorenzMeier, 2015-10-30:
@DonLakeFlyer Waypoints are only displayed if you click on load, despite that the vehicle shows that they have been loaded on the connection on boot.

DonLakeFlyer, 2015-10-30:
I can't repro this on very latest master. Maybe it is flaky somehow?

LorenzMeier, 2015-10-30:
Retrying.

LorenzMeier, 2015-10-30:
No, confirmed. It only displays once I manually click load. It loads the things before though, as I can see from the PX4 transfer log.

LorenzMeier, 2015-10-30:
To repro:

1) Go to fly
2) Connect the vehicle
3) Go to plan

In a few cases there will be no mission in the plan view.

DonLakeFlyer, 2015-10-30:
I tried it with my Iris and can't get a repro

DonLakeFlyer, 2015-10-30:
At first I thought it wasn't working in the Fly view but my mission was just off screen.

closed 2015-12-03 by DonLakeFlyer