mavlink/qgroundcontrol #2172: AutoConnect ToDo List

DonLakeFlyer, 2015-10-31:
Here are the known remaining work for UDP AutoConnect:
- Either find a new way to manage link counts, or find all the code that depends on link count to work and change it. Known places: Connect/Disconnect button, Firmware Upgrade, Log Replay, ...
- Figure out how to handle Airframe Config Apply/Restart. The vehicle timeout will not catch this case
- Mavlink logging needs to terminate on all vehicles gone, not all links gone
- Need to find all the places in the code which watches for link disconnect, and instead watch for Vehicle disconnect
- Still not sure if we need to catch the time going backwards to detect a vehicle reboot and re-query parameters/missions. Vehicle timeout disconnect catches some of this but not all.

Still working on this...

LorenzMeier, 2015-10-31:
Let me know if I can help with specific questions.

DonLakeFlyer, 2015-10-31:
I think I have answers for everything. It's just work at this point.