mavlink/qgroundcontrol #2182: Mission editing from Fly view

DonLakeFlyer, 2015-11-03:
Mission editing from Fly view
- Plan view goes away
- Two modes of editing from Fly view, full/limited
- Full mode has all currently available functionality from current Plan screen
- Limited mode is a subset  of editing capability available while the vehicle is actively flying 
- Reason for limited mode is to keep visuals while active flight uncluttered and focused on flying
- Reason for limited is to disallow possibly unsafe changes to mission while actively flying vehicle 
- Functionality available during limited mode is TBD
- Likely candidates: waypoint move, set next mission item, change altitude
- Pausing the vehicle during flight will switch to full editing mode
- Pause mode of fly view will dial back Fly visuals to only what is needed to verify nothing isn't going wrong during loiter
- If there is no mission on the vehicle pause mode has no effect on Fly visuals 
- Pause mode in turn clears space and sets user focus to mission editing environment while vehicle waits for you to finish detailed editing
- Full editing available prior to flight
- Arm vehicle transitions to limited mode editing

dagar, 2016-05-28:
Is this still on the table? I know a few people that really dislike the separate Plan/Fly.
Another frequent request is tweaking the position of a waypoint during a mission.

DonLakeFlyer, 2016-05-28:
At some point. But way in the future. You should enter tweaking a waypoint as a seperate issue.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier