mavlink/qgroundcontrol #2192: Try Qt 5.6 beta with Android 5+

LorenzMeier, 2015-11-06:
This hopefully fixes the black screen issue. It also add Qt Positioning (GPS) for Android and iOS, which would allow follow-me.

dogmaphobic, 2015-11-07:
> It also add Qt Positioning (GPS) for Android

This has been working for me with the code Don added a while back (all versions of Android I have and Qt 5.5.1)

LorenzMeier, 2015-11-07:
I would be interested to experiment with follow-me with that. We should send this message with the GPS coordinates: http://mavlink.org/messages/common#GLOBAL_POSITION_INT

Once we have this I can add support in the Firmware to follow this on a leash. Should be trivial.

dogmaphobic, 2015-11-07:
I'll wait for Don for that one. The current code gets the local coordinates just once to establish your starting position. It's trivial to get the coordinates continuously though. Get it down the pipe and out to the current vehicle is something Don could probably do a lot faster than I would.

thedevleon, 2016-05-18:
Any updates on this? I would like to use follow me with QGroundControl :)

DonLakeFlyer, 2017-01-12:
Because it not implemented: #4427

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier