mavlink/qgroundcontrol #2204: Loading waypoint list is desperatelly slow.

Fojtik, 2015-11-12:
Pleas try to load my textual file "AroundCech.waypoints". It takes more than 15 minutes to load it on my computer. There is something very unclean. Every single waypoint enforces to render everything anew.

DonLakeFlyer, 2015-11-15:
I tried loading this file using the daily build. The first 4 waypoints have a command of 0 which is invalid. So the daily build refuses to show waypoints for this file. I tried changing the first 4 items to command = 16. Then I ran into the problem that some of the waypoints have auto continue set to false. Which also prevents the waypoints from being displayed.

Fojtik, 2015-11-16:
I have generated this file with QGC. 

I have set relative altitude for these 4 waypoints, it is the only difference.

DonLakeFlyer, 2015-11-16:
> I have generated this file with QGC.

Were you using the Daily build or Stable? Mission planning is completely different code in Daily. I assume you are running into save bugs in Stable which probably no longer exist in Daily.

Fojtik, 2015-11-16:
Dear Don,

> > I have generated this file with QGC.
> 
> Were you using the Daily build or Stable? Mission planning is completely different code
> in Daily. I assume you are running into save bugs in Stable which probably no longer
> exist in Daily.
I have no clue about save bug. I am using stable build for regular testing.

It seems to me that daily build is quite limited. Yes, there is a new waypoint editor,
but the old waipoint editor seems better for some purposes. It has table with waypoints 
that could be arbitrary modified. The daily build behaves somehow different.

regards
   Jara

DonLakeFlyer, 2015-11-16:
> It has table with waypoints that could be arbitrary modified.

The waypoint list is not on the right hand side of the screen. The big difference is that it doesn't support things for visual editing which the current PX4 Firmware does not support. Whereas old editor supported things that didn't actually work. What other issues are there?

I'll try to hack up your file again to test performance.

DonLakeFlyer, 2015-11-16:
Before QGC will show a vehicle position it must receive a MAVLINK_MSG_ID_GPS_RAW_INT message from the vehicle with the fix_type field of the message set to 3 or higher which indicates a 3d fix or higher.

I don' t understand what you mean by "Other". There is no MAV_CMD for a mission item which is "Other". Did old QGC support this, if so what MAV_CMD was it sending?

Fojtik, 2015-11-16:
Dear Don,

> Before QGC will show a vehicle position it must receive a MAVLINK_MSG_ID_GPS_RAW_INT
> message from the vehicle with the fix_type field of the message set to 3 or higher
> which indicates a 3d fix or higher.
Thanks for advice. This message was not required till now. Everything worked fine.

> I don' t understand what you mean by "Other". There is no MAV_CMD for a mission item
> which is "Other". Did old QGC support this, if so what MAV_CMD was it sending?
Please have a look at the image attached.

For my vehicle it is best to provide several generic fields that are depicted
on the image.

The WAYPOINT has latitude, longitude, altitude and several binary flags that can be 
packed to some "free" field. I can do any translation for QGC, but the attributes 
obtained from vehicle should be resent during update.

regards
   Jara

DonLakeFlyer, 2015-11-16:
> Please have a look at the image attached.

There is no image to look at

Fojtik, 2015-11-16:
The image has been sent to your Email. I am 100% sure. I have attached image to previous comment manually.

dogmaphobic, 2015-11-16:
> [image]

The image is just a link. He's been using email instead of posting directly. That's why it didn't show up.

DonLakeFlyer, 2015-11-16:
Ah interesting. I could add a similar thing that just exposes all item settings. The Cmd = 0 is very strange though since that isn't a supported MAV_CMD.

@Fojtik Can you add a separate Issue for adding an Other or something like that item? And attach this image to that Issue?

Fojtik, 2015-11-16:
Dear Don,

> Ah interesting. I could add a similar thing that just exposes all item settings. The
> Cmd = 0 is very strange though since that isn't a supported MAV_CMD.

I really don't know which CMD to use. I have used something that proves to work.

But it suddenly gets broken with new version and no replacement known.

 
> @Fojtik Can you add a separate Issue for adding an Other or something like that item? And attach this image to that Issue?
Yes, we are out of this topic ;).

regards
  Jara