mavlink/qgroundcontrol #2207: Delete points in QGC crashed

Fojtik, 2015-11-13:
I am using release 2.7.1 Delete points in QGC crashed. Here is a callstack.

>	Qt5Core.dll!671b3da6() 	
 	[Frames below may be incorrect and/or missing, no symbols loaded for Qt5Core.dll]	
 	msvcr120.dll!00d7ecfa() 	
 	qgroundcontrol.exe!mapcontrol::TrailItem::`scalar deleting destructor'()  + 0x55 bytes	C++
 	qgroundcontrol.exe!mapcontrol::UAVItem::DeleteTrail()  Line 210 + 0xc bytes	C++
 	qgroundcontrol.exe!QGCMapWidget::deleteTrails()  Line 121 + 0x7 bytes	C++
 	Qt5Core.dll!671b4d2f() 	
 	Qt5Core.dll!671b4f4e() 	
 	Qt5Widgets.dll!6529dd34() 	
 	Qt5Widgets.dll!650bec04() 	
 	Qt5Widgets.dll!650beaf8() 	
 	Qt5Widgets.dll!650bf67e() 	
 	Qt5Widgets.dll!65032d38() 	
 	Qt5Widgets.dll!6500ec77() 	
 	Qt5Gui.dll!01164194() 	
 	Qt5Widgets.dll!650ac6ab() 	
 	Qt5Core.dll!671955ec() 	
 	Qt5Core.dll!67211ce1() 	
 	Qt5Widgets.dll!650136a9() 	
 	Qt5Widgets.dll!65050902() 	
 	Qt5Widgets.dll!6504f3a3() 	
 	Qt5Widgets.dll!6504af79() 	
 	Qt5Widgets.dll!650125d7() 	
 	Qt5Widgets.dll!650125e9() 	
 	Qt5Widgets.dll!650107a7() 	
 	Qt5Gui.dll!0115103d() 	
 	qwindows.dll!026ebde6() 	
 	Qt5Core.dll!671d578f() 	
 	user32.dll!7e37b372() 	
 	user32.dll!7e368734() 	
 	user32.dll!7e368816() 	
 	user32.dll!7e3689cd() 	
 	Qt5Core.dll!671902d8() 	
 	Qt5Core.dll!671d51a7() 	
 	ntdll.dll!7c910435() 	
 	ntdll.dll!7c91019b() 	
 	ntdll.dll!7c910041() 	
 	ntdll.dll!7c9142b9() 	
 	ntdll.dll!7c913f92() 	
 	ntdll.dll!7c914217() 	
 	ntdll.dll!7c91019b() 	
 	ntdll.dll!7c910041() 	
 	ntdll.dll!7c91005d() 	
 	ntdll.dll!7c91005d() 	
 	msvcr120.dll!00d7ed63() 	
 	msvcr120.dll!00d7ecfa() 	
 	Qt5Core.dll!67029cb3() 	
 	Qt5Core.dll!6714700f() 	
 	Qt5Core.dll!671833aa() 	
 	msvcr120.dll!00d7ed63() 	
 	Qt5Core.dll!67029c3c() 	
 	qwindows.dll!026ebd2e() 	
 	Qt5Core.dll!67190fef() 	
 	Qt5Core.dll!67192f70() 	
 	Qt5Gui.dll!0114d9f9() 	
 	Qt5Gui.dll!0114da0f() 	
 	Qt5Gui.dll!0114dae7() 	
 	Qt5Gui.dll!0114b909() 	
 	qgroundcontrol.exe!main(int argc=0x00000001, char * * argv=0x00177078)  Line 198 + 0x6 bytes	C++
 	qgroundcontrol.exe!__tmainCRTStartup()  Line 626 + 0x17 bytes	C
 	kernel32.dll!7c816037() 	
 	qgroundcontrol.exe!std::swap<QextPortInfo>(QextPortInfo & _Left={...}, QextPortInfo & _Right={...})  Line 52 + 0x6c bytes	C++
 	20202020()

DonLakeFlyer, 2015-11-13:
> qgroundcontrol.exe!std::swap(QextPortInfo & _Left={...}, QextPortInfo & _Right={...}) Line 52 + 0x6c bytes C++

The Daily Builds use completely different code for waypoint editing and map display than the last Stable build. So this problem is likely no longer there in the latest builds.

Fojtik, 2015-11-13:
Dear Don,

> The Daily Builds use completely different code for waypoint editing and map display
> than the last Stable build. So this problem is likely no longer there in the latest
> builds.

I cannot fully run daily build. May be this is a problem of Czech Windows. It runs,
but does not display map.

D:\Program Files\qgroundcontrol>qgroundcontrol.exe
[I] at :0 - "Logging ini file directory "H:/Documents and Settings/Administrator.DVRMN/Local Settings/Data aplikac+s/QtProject""
[I] at :0 - "Filter rules "qt.network.ssl.warning=false\n""
[I] at :0 - "Settings location "H:/Documents and Settings/Administrator.DVRMN/Data aplikac+s/QGroundControl.org/QGroundControl.ini" true"
[I] at :0 - "Saved files location "H:/Documents and Settings/Administrator.DVRMN/Dokumenty/QGroundControl""
[E] at qrc:qml/MainWindow.qml:129 - "qrc:qml/MainWindow.qml:129:5: Type MainToolBar unavailable"
[E] at qrc:///qml/QGroundControl/Controls/MainToolBar.qml:430 - "qrc:///qml/QGroundControl/Controls/MainToolBar.qml:430:9: Type QGCButton unavailable"
[E] at qrc:///qml/QGroundControl/Controls/QGCButton.qml:9 - "qrc:///qml/QGroundControl/Controls/QGCButton.qml:9:1: Type Button unavailable"
[E] at file:///D:/Program Files/qgroundcontrol/QtQuick/Controls/Button.qml:70 - "file:///D:/Program Files/qgroundcontrol/QtQuick/Controls/Button.qml:70:1: Type
Menu unavailable
     BasicButton {
     ^"
[E] at file:///D:/Program Files/qgroundcontrol/QtQuick/Controls/Menu.qml:150 - "file:///D:/Program Files/qgroundcontrol/QtQuick/Controls/Menu.qml:150:5: Cannot
assign to non-existent property "on__MenuClosed"
         on__MenuClosed: __currentIndex = -1
         ^"
[I] at :0 - "QGCMessageBox (unit testing) "Qml Error" "Source not ready: Status(3)\nErrors:\nqrc:qml/MainWindow.qml:129:5: Type MainToolBar unavailable\nqrc:///
qml/QGroundControl/Controls/MainToolBar.qml:430:9: Type QGCButton unavailable\nqrc:///qml/QGroundControl/Controls/QGCButton.qml:9:1: Type Button unavailable\nfi
le:///D:/Program Files/qgroundcontrol/QtQuick/Controls/Button.qml:70:1: Type Menu unavailable\nfile:///D:/Program Files/qgroundcontrol/QtQuick/Controls/Menu.qml
:150:5: Cannot assign to non-existent property \"on__MenuClosed\"\n""
[I] at :0 - "0"
[I] at :0 - "Orphaned log file count 0"
[I] at :0 - "After app delete"

Here I see deffect
   "Data aplikac+s"
There should be:
   "Data aplikací"

regards
  Jara

dogmaphobic, 2015-11-13:
> "file:///D:/Program Files/qgroundcontrol/QtQuick/Controls/Menu.qml:150:5: Cannot
> assign to non-existent property "on__MenuClosed"
>          on__MenuClosed: __currentIndex = -1
>          ^"

That indicates that the Qt version you have in ```D:/Program Files/qgroundcontrol/QtQuick/Controls/Menu.qml``` is 5.4.2. The current QGC only runs with Qt 5.5.1

Fojtik, 2015-11-14:
Dear Gus,

> That indicates that the Qt version you have in ```D:/Program Files/qgroundcontrol/QtQuick/Controls/Menu.qml``` is 5.4.2. The current QGC only runs with Qt 5.5.1
Yes, you are right. The problem behind is that your automatic installer does not
overwrite and update QT. When I completelly erased installation and reinstall,
it started to work.

----

Yes, the new version does not crash. But it behaves completely different.
When I download waypoints from my system they are not displayed at all.

regards
   Jara

closed 2015-11-15 by DonLakeFlyer