mavlink/qgroundcontrol #2208: UAVCAN ESC output ID needed for Fixed Wing

pkocmoud, 2015-11-13:
I am using a fixed wing setup. I have several ESCs which have been enumerated. They are ID'd as Component #125 and #124. Under Settings:Power:UANCAN Bus Configuration, there is a check box for "Enable UAVCAN as the default MAIN output bus". I think what is missing is a way to clearly define which CAN Component ID it will output to. When connected separately, the 125 will spin the motor and the 124 does not.

proficnc, 2016-07-28:
Trying to do the same thing.... 
under parameters, only Component #1 is showing up.
and there is no obvious way to configure the esc's

DonLakeFlyer, 2016-07-28:
If the vehicle sends the parameters to QGC it will show them even if from a different component. YOu can see this working with the WiFi modfule for example. If QGC isn't showing a component's parameter then it is very likely the vehicle never sent them.

proficnc, 2016-07-28:
I can load QGC 2.3 and they show, but with latest they don't. 

This has been replicated by Ben Dyer.

DonLakeFlyer, 2016-07-29:
Strange. Yeah, you're going to need to send me some of these so I can test. I'll ping Phillip.

DonLakeFlyer, 2016-08-18:
Ok, can someone send me one of these to test with. Do I need more than one for bench testing?

proficnc, 2016-08-19:
Will do, as soon as I can get one :)

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier