mavlink/qgroundcontrol #2209: UAVCAN ESC Motor Index and Direction Assignment - Clarification

pkocmoud, 2015-11-13:
This should be a toggle button as it is not clear which mode it is in. I notice in NSH that when an ESC is assigned there is output to the console, I feel there needs to be something displayed on this QGC screen as well. There should probably have a timeout to switch from Start to Stop.

proficnc, 2016-07-28:
agreed.

DonLakeFlyer, 2016-07-28:
I'm going to need way more detail than that since I know very little about uavcan. QGC only has available to it parameters which it can change/show and information coming to it over mavlink. Can you relate this back to a certainly parameter of mavlink information.

proficnc, 2016-07-28:
Sounds like we need to get you some Can devices :) 

Ping me or Phill K, I'm sure we can help out (Skype is best for me...

closed 2021-01-22 by LorenzMeier