mavlink/qgroundcontrol #2212: MAVLINK gimbal control via PX4 flight stack

estechnical, 2015-11-14:
Is it possible to configure/control a mavlink capable gimbal (eg STorM32 BGC based gimbals) via qgroundcontrol? I would like to attach my gimbal to the Pixhawk on a telem/serial port and be able to use qgroundcontrol to select preset camera positions for example.

DonLakeFlyer, 2015-11-14:
Depend on what it takes to configure. It it's just parameter setting then it should be doable. Not sure what you mean by preset camera positions?

estechnical, 2015-11-14:
I have a 3 axis gimbal that's operated by a second TX (providing ppm into the main radio and mixing etc) and the gimbal is capable of emitting a mavlink heartbeat but I've not yet got this connected up in such a way that I've been able to see the mavlink data from it in qgroundcontrol.

The gimbal setup tool has a cool little widget that allows you to recentre any axis or the whole gimbal, and it has positions like 'look down' and 'look forwards'. This nicely compliments the relative positioning control of the gimbal we have via the 2nd tx if it is possible...

estechnical, 2015-11-15:
I put a message in the PX4 user group too, see https://groups.google.com/forum/#!topic/px4users/nhp1LsaQxbM

I envisage there being a tool widget that has a range of buttons:
- look down
- look down at 45 degrees
- look forwards
- center roll
- center pitch
- center yaw
- center all axes

and possibly a couple of buttons that can be configured for gimbal positions of your own invention.

DonLakeFlyer, 2015-11-15:
If there are controller by mavlink COMMAND_LONG messages then you can write a custom command widget to control it: http://www.qgroundcontrol.org/custom_command_qml_widgets

estechnical, 2015-11-15:
Please could you take a look at this: http://www.olliw.eu/storm32bgc-wiki/Serial_Communication#Mavlink_Communication and let me know if this is something that should be compatible? I'd probably go for this option if it would work...

Thanks :)

Ed

DonLakeFlyer, 2015-11-15:
You can implement your own UI using a custom command widget for anything supported by 
MAVLINK_MSG_ID_COMMAND_LONG

Molecuman, 2017-07-06:
Why this is more difficult than Mission Planner? Is there an easy way to do this?

DonLakeFlyer, 2017-07-06:
Gimbal control is now more easily support in missions (daily build) but still not gimbal controls to use while just flying manually.

> Why this is more difficult than Mission Planner?

Simply because the feature is not yet implemented.

Molecuman, 2017-07-18:
> Gimbal control is now more easily support in missions (daily build) but still not gimbal controls to use while just flying manually.

So at this moment there is no way to control a Mavlink Storm32 gimbal thru manual controls on the radio transmitter?
I try to search a step by step instructions to setup the serial port for controlling the Storm32 but with no success. Do anybody know how can I do this? I´m using the Arducopter stack, is more easy to do it but I want to test the PX4 stack. Thanks!

DonLakeFlyer, 2017-07-18:
> So at this moment there is no way to control a Mavlink Storm32 gimbal thru manual controls on the radio transmitter?

You need to ask firmware folks about that.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier

LorenzMeier, 2021-01-22:
And it is implemented fully by now.