mavlink/qgroundcontrol #2216: Show reasonable default location, link default location for edit and fly view

LorenzMeier, 2015-11-14:
The edit view shows this by default. You have to zoom out quite a bit before you see something dark blue, and you have to zoom out very far to figure out you're in the sea south of west africa.

It does confuse users like hell:
https://groups.google.com/d/msg/qgroundcontrol/1-VRXDbI49Q/wiEfg1LWDgAJ

[image]

DonLakeFlyer, 2015-11-14:
This is with current daily? It should runs through a set of possible positions depending on tablet/home valid and so forth. I'll take a look. Desktop not connected should be ETH position. Tablet not connected should be tablet position.

LorenzMeier, 2015-11-14:
Note: It looks the same on my devices. This is *not* user or OS specific.

LorenzMeier, 2015-11-14:
Yes, current daily and master are affected the same way.

DonLakeFlyer, 2015-11-14:
I can't repro this. It's working correctly on desktop for me. Did you just boot and go to edit or do something else in between?

dogmaphobic, 2015-11-14:
I've been seen this for quite some time. The difference is that I knew what this was. We used to have google as the default provider and that one would show the Atlantic Ocean at 0,0. Bing doesn't have those tiles so it shows the image above.

I assumed it was something Don was still working on and wasn't quite ready. Note that this 0,0 behavior was there even before Don added the new mission code. The only difference is that Bing is now the default provider.

The fix is to not ever have 0,0 as the default coordinate.

DonLakeFlyer, 2015-11-14:
Neither map has defaulted to 0,0 coordinate for weeks now. Both of them have a set of defaults they use based on what is available. It was recently broken somehow.

LorenzMeier, 2015-11-14:
Don, you likely have a cached location somewhere. I agree with Gus that the issue is defaulting to 0,0. And both views should default to the *same* and be linked in terms of center and zoom level by default. Its super annoying to have to re-focus the view individually when you switch between them for anyone trying to plan and follow a real mission.

It might need some soft constraints as in setting the edit location to the vehicle location but not letting it track the vehicle like the flight view once it has the vehicle in view or something like that.

I would try to not get carried away with logic but mostly focus on making both views initialise to the same locations on boot and on vehicle connect.

DonLakeFlyer, 2015-11-14:
So you are telling me that all the code I wrote to default to tablet location if available, home position if available, or track vehicle location if available, or default to a non-zero location never existed! This is a bug. It used to all work correctly. There is no such thing as a cache location. Neither map does that.

LorenzMeier, 2015-11-14:
No, I mostly tried to point out the confusion arising from the non-linked maps, as it seemed that adding that as feature might be not much more effort when fixing this one.

Is the defaulting code maybe enabled for fly and not for edit? I recall having seen 0,0 for quite some time in edit, while fly worked great.

LorenzMeier, 2015-11-15:
Within the tablet scope fixed.

closed 2015-11-15 by LorenzMeier