mavlink/qgroundcontrol #2221: Vendorid returned by QSerialPortInfo.vendorIdentifier() incorrect

thomasgubler, 2015-11-15:
On my system this started to return 0x8086 instead of 0x26ac which belongs to intel [1] and also corresponds to the root node in the usb device tree (see [3]). I have to manually patch SerialPortIds.h [2] to make qgc discover my pixhawk board. 

Unfortunately I can not pin the problem to a specific upgrade.

```
qmake --version
QMake version 3.0
Using Qt version 5.5.1 in /usr/lib
```

[1] http://pcidatabase.com/vendor_details.php?id=1302
[2] https://github.com/mavlink/qgroundcontrol/blob/master/src/SerialPortIds.h#L31
[3] http://pastebin.com/qK8xcEu0

LorenzMeier, 2015-11-15:
@DonLakeFlyer We might have to fall back to string matching instead.

thomasgubler, 2015-11-15:
The same issue occurs with radio modules which is pretty annoying since by simply replacing the vendorids a usb-connected pixhawk can not be distinguished from the radio module and the wrong settings are applied to the port.

DonLakeFlyer, 2015-11-15:
This pull #2218 standardardized the port finding code. Flash code has good string comparison fallback, but I don't think connect did. They will both do the same thing now. Currently the pull has no Radio string matching fallback. I'll see if that is possible.

DonLakeFlyer, 2015-11-15:
@thomasgubler Do you get good port descriptions through or are those screwed up as well? If not, there will be no way to to string compare fallback. Is it just Qt that is returning bad values, or are they coming from the OS. If from the OS, I'd think there is an OS problem since returning the incorrect ids will likely screw up lots of folks code.

Once #2218 passes CI can you give it a try to see if both fallbacks work?

thomasgubler, 2015-11-16:
#2218 fixes this. The descriptions are coming through and it's using the name matching fallback.

cool, thanks for the quick response!

closed 2015-11-16 by thomasgubler