mavlink/qgroundcontrol #2227: Mission editor missing parameters for waypoints

edkoch, 2015-11-16:
I can't find a way to edit all the parameters for the waypoints in a mission (e.g. heading). I saw some reference to there being a simple and full editing mode for missions so if the missing parameters can be accessed by changing modes then how do you do that?

DonLakeFlyer, 2015-11-16:
Heading is not currently exposed to the user since most firmwares default to controlling the heading themselves during a mission. We'll need to create an autopilot plugin check to determine when that is not the case, and then show heading.

edkoch, 2015-11-16:
I suppose it depends upon wether we are talking about multirotor or fixed wing. For multirotor the following parameter dictates how the heading parameter for waypoints are handled:

MIS_YAWMODE	Multirotor only. Yaw setpoint mode	0	3	1
0: Set the yaw heading to the yaw value specified for the destination waypoint. 1: Maintain a yaw heading pointing towards the next waypoint. 2: Maintain a yaw heading that always points to the home location. 3: Maintain a yaw heading that always points away from the home location (ie: back always faces home). The values are defined in the enum mission_altitude_mode

edkoch, 2015-11-16:
Don,

BTW it isn’t just the heading of the waypoint I have had to edit in the past, but also the uncertainty radius at the waypoint which also is not exposed.

-ed

DonLakeFlyer, 2015-11-16:
Do you think needing uncertainty radius is a fixed wing thing as well?

edkoch, 2015-11-16:
I don’t fly fixed wing so I don’t know, but it is probably unlikely. One thing to note is that even though I have had to edit the uncertainly radius in the past I have always used the same value for all my waypoints. I’ve noticed a couple of other things I use to be able to do, but no longer can.  I’m in the process of sending you an email with my use case described for how I edit my missions. You should receive that shortly.

-ed

edkoch, 2015-11-16:
Don,

I think that your new UI is looking very good, but now that I have had a chance to play with the mission editor some more let me describe my use case and what I am no longer able to do that I could before.

I typically define my mission file offline using a variety of tools in a semi-automated fashion, i.e. I typically don’t define all my waypoints by hand. The mission file I create in this fashion typically does not have a takeoff and landing point in the file. I go out into the field and then decide where my takeoff and landing points should be based upon the conditions. I then load the mission file into QGC and then by hand click on the map to define the takeoff and landing positions. On occasion I might also add new waypoints into the middle of the mission to circumvent some unanticipated obstacle. The main point is that with the old QGC I could insert waypoints wherever I wanted, but with the new one I can only add them to the end. Obviously we need the ability to insert new waypoints anywhere we want. BTW in the old QGC it is VERY painful to insert new waypoints into the middle of a mission since you first added them at the end and then used an up arrow to move them up the list. Adding a takeoff point was especially painful if you had a lot of waypoints. I’m hopeful that we can improve on that.

thanks,

-ed

DonLakeFlyer, 2015-11-16:
Hmm. I had Move Up/Down in there for a while but removed it.  Would have suffered from the same issue as the old editor. Let me think about some way to do this in s user friendly manner.

DonLakeFlyer, 2015-11-23:
https://github.com/mavlink/qgroundcontrol/pull/2261

DonLakeFlyer, 2015-11-30:
FYI: "Raw Editing" is now supported for all mission items. Click the little cog wheel icon and you have access to all params for the item.

So work still left is the ability to move around mission items.

DonLakeFlyer, 2016-01-21:
We now have delete and insert. Move is not going to happen until a Qt bug is fixed which prevents us from doing item moves.

closed 2016-01-21 by DonLakeFlyer