mavlink/qgroundcontrol #2229: Implement waypoint with position and generic attributes

Fojtik, 2015-11-16:
Thanks for advice. This message was not required till now. Everything worked fine.
 I don' t understand what you mean by "Other". There is no MAV_CMD for a mission item
 which is "Other". Did old QGC support this, if so what MAV_CMD was it sending?
Please have a look at the image attached.

For my vehicle it is best to provide several generic fields that are depicted
on the image.

The WAYPOINT has latitude, longitude, altitude, speed and several binary flags that can be
packed to some "free" field. I can do any translation for QGC, but the attributes
obtained from vehicle should be resent during update.

DonLakeFlyer, 2015-11-23:
https://github.com/mavlink/qgroundcontrol/pull/2261

DonLakeFlyer, 2015-12-12:
Full editing of all parameters is now supported in raw edit mode. Click the gear icon to get into raw edit. But you have to use one of the existing MAV_CMD ids from the mavlink headers. There is a separate issue on the mavlink side which says that there should be some _USER cmd ids you could use. But that change needs to happen in the mavlink repro.

closed 2015-12-12 by DonLakeFlyer