mavlink/qgroundcontrol #2240: Number formats not too user friendly

dogmaphobic, 2015-11-19:
Some numbers are converted to not too friendly formats. For instance, if you enter 5200 (mA) for a battery, it gets converted to:
[image]
Some other numbers are shown with way too many digits of precision:
[image]
Longitude and Latitude are especially too complicated:
[image]

LorenzMeier, 2015-11-19:
I presume you're aware we introduced the decimal field in the param meta data for that, but its not propagated completely yet?

As for lat and lon: You're missing that this full precision is required. Any less than that and you start to raster by a couple dozen meters.

dogmaphobic, 2015-11-19:
> I presume you're aware we introduced the decimal field in the param meta data for that, but its not propagated completely yet?

No, I was not.

> As for lat and lon: You're missing that this full precision is required. Any less than that and you start to raster by a couple dozen meters.

It's not a matter of precision. It's about human readability. A latitude of 64.154549 is a heck lot easier to understand than 641545490*1e7. The first is immediately obvious it is way the heck up north. The other could be either way up north or near the equator. I have to carefully count the digits and make the math to figure it out.

Especially confusing for battery mA. It's a lot simpler to understand 5200mA than 5.2+03mA. Might as well use 5.2A instead.

The default value above is obviously a floating point rounding error. In other words, where the need of precision is required, use straight decimals. Not exponents. When 4 decimal points of precision is enough, limit it to 4, etc.

It's all about minimizing the chances of user error.

DonLakeFlyer, 2015-11-19:
BAT_CAPACITY is a format string problem in the FactSystem.
Default Value (and min/max) display should use the same float precision as the parameter value uses.
LAT/LON is working correctly and goes through no conversion. Parameter is specified as an int.

DonLakeFlyer, 2015-11-20:
This is fixed. Lat/Lon is working correct per the parameter type. If you want something other than that it's a firmware change.

closed 2015-11-20 by DonLakeFlyer