mavlink/qgroundcontrol #2241: Add back true map caching

LorenzMeier, 2015-11-19:
The current approach with zooming in and out only works for a small flying field and even there not reliably. We need a way to load all tiles in the current view down to the max. zoom level on button click.

@dogmaphobic How hard / easy is that with our map backend?

dogmaphobic, 2015-11-19:
As we discussed earlier in the wrong place :) there are two ways to go about this. The hacky way is to programmatically zoom and pan the map around waiting for QtLocation to cache the tiles on its own. I don't like it because it's wasted development time. The proper way is to implement our own caching. That way I can launch *n* threads collecting tiles in bulk. It would be orders of magnitude faster. For either scenario, the part that will take the longest is the UI.

I had already written the caching code. I just need to find it...

DonLakeFlyer, 2015-11-19:
> That way I can launch n threads collecting tiles in bulk.

Cool!

dogmaphobic, 2015-12-22:
We might have a problem with this. For that matter, for all of it. According to article *2.c* under **Prohibited Conduct**, it states:

"*mass download or create bulk feeds of the Content (or let anyone else do so);*"

http://maps.google.com/help/terms_maps.html

Bing maps has the same sort of language, though it uses 1,000,000 words to say the same thing, with one caveat: They do have special treatment for *non profit*:

http://www.microsoft.com/maps/product/terms.html

For tile fetching, both assume we are using an *appropriately authorized* API for accessing the map tiles. We don't. We fetch tiles the same way a web browser would fetch them if you were browsing google maps or bing maps. We even provide a *user agent* presenting ourselves as a full fledged browser.

So far, when we present tiles on the Fly/Mission view, only a relatively small number of requests are sent. Even if you were to pan around a lot, it would be not that different were you doing it from a web browser and it would not generate an absurd number of requests. However, allowing people to define a *region* and collect all tiles in that region for a *range* of zoom levels could easily translate into millions of requests, which clearly goes against their respective terms of use. Google does not allow it, period. Microsoft has an allowance of around 10,000 or 50,000 requests (could not tell which is which) per 24-hour period (for a *registered* application using an authorization key).

Here is a conversation between google and an open source project that (used to) provide google map caching for offline use. They no longer use google because of this. The important bits are the letters from google, not the ignorant ramblings in between:

https://code.google.com/p/gmapcatcher/issues/detail?id=210

Not sure how to proceed...

dogmaphobic, 2015-12-22:
By the way, the map fetching code in QGC is loosely derived from the OpenPilot project. Not naming names, this same code is used by quite a few other GCS projects.

Nyaruko, 2015-12-22:
But at least the osm is available?

NaterGator, 2015-12-22:
Perhaps one option is to let users specify a path to their own local repository of maps? The work would be in demonstrating legal means of acquiring Qt compatible tilesets or writing a QtLocation plugin to render standard map sources like geotiffs.

dogmaphobic, 2016-05-06:
It's there...

closed 2016-05-06 by dogmaphobic

brdgtannj, 2016-11-22:
Someone prior mentioned letting users specify a path to their own repository of maps or using a QtLocation plugin to render sources such as geotiffs.  You mentioned "it's there" and I'm curious if you'd elaborate?  I'm trying to integrate a geotiff and any suggestions/advice would be really helpful!

DonLakeFlyer, 2016-11-22:
I'd ask dogmaphobic on gitter, referencing this issue.