mavlink/qgroundcontrol #2242: Problems to connect over UDP

Shokman, 2015-11-19:
I am experiencing troubles since 2 weeks ago to connect to the SITL code from qgroundcontrol. I am connecting through MAVROS. Mavros communication works flawless. But when I try to connect to the vehicle through mavros gcs_url settings I don't get the full parameter list or it disconnects while asking the waypoints list. This was working ok until we needed to change Qt library as it is described here #2075, two weeks ago. 

Maybe is related with issue #2176? 

I would like to provide further information, for debug log I need to compile myself the code?

DonLakeFlyer, 2015-11-19:
I doubt this has anything to do with change Qt versions. This is more likely caused by the recent change to disconnect a vehicle if communication is lost on the link for more than 30 seconds. So either mavros has stopped communicating with QGC or QGC is somehow losing the packets from mavros.

I’ve never used mavros. If it’s simple enough to setup and you can detail exactly the steps to repro this I can try to reproduce it myself.

Shokman, 2015-11-19:
This is my mavros launchfile

```
<launch>
<arg name="ns" default="/" />

<group ns="$(arg ns)">
  <arg name="fcu_url" default="udp://localhost:14550@localhost:14556" />
  <arg name="gcs_url" default="udp://localhost:14574@localhost:14576" />
  <arg name="tgt_system" default="1" />
  <arg name="tgt_component" default="1" />

  <arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
  <arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

  <node pkg="mavros" type="mavros_node" name="mavros" clear_params="true" output="screen">
    <param name="fcu_url" value="$(arg fcu_url)" />
    <param name="gcs_url" value="$(arg gcs_url)" />
    <param name="target_system_id" value="$(arg tgt_system)" />
    <param name="target_component_id" value="$(arg tgt_component)" />

    <!-- load blacklist, config -->
    <rosparam command="load" file="$(arg pluginlists_yaml)" />
    <rosparam command="load" file="$(arg config_yaml)" />
  </node>
</group>
</launch>
```

I've been always working with it, but now mavros bridge/qgroundcontrol settings.
You can install mavros following this: https://github.com/mavlink/mavros/blob/master/mavros/README.md
(you are going to need ROS). Then I connect to listening port 14576 and to target host on 14574 as UDP connection.

You should use a SITL simulation also, I am using a parallel project to the gazebo sitl configuration (https://github.com/PX4/sitl_gazebo). It is practically the same, but we use our custom environment and gazebo plugins. I didn't test it using traditional JMavSim. Maybe @tumbili who has been working with that knows something about this?

It is weird, because disconnection occurs before than 30 seconds connection and I know that packages are received, but it never survives the waypoint list request... sometimes I get the connection paramaeter part almost finished, and then it disconnects.

I told @vooon  about this on the gitter channel, maybe it should be listed as a mavros issue? I've used qgroundcontrol + mavros for a long time (like one year) without any issue, and this configuration worked for months. Over serial is working ok. Maybe there is an issue with HIL messages used with SITL setup?

DonLakeFlyer, 2015-11-19:
Hard to say whether it’s macros, SITL or QGC at this point.

Shokman, 2015-11-19:
Do we have some qgroundcontrol verbose mode? I am running it from the terminal, but there is no error displayed.

DonLakeFlyer, 2015-11-19:
Yes, there is but I don’t think the disconnect is logged. I’ll update that.

vooon, 2015-11-19:
@Shokman your url's little strange: `udp://localhost:14550@localhost:14556`, usually 14550 is a QGC bind port.

And check mavros logs for remote ip changes. Maybe some process changes that?

@DonLakeFlyer how about on doing multicast UDP? In that case we may omit those bridges completely.

vooon, 2015-11-19:
@Shokman i think `wireshark` is best tool to check who feeding data to 14576 port.

DonLakeFlyer, 2015-11-19:
> how about on doing multicast UDP? In that case we may omit those bridges completely.

You'd have to talk with @dogmaphobic about that. The only think I know technically about UDP is how to spell it!

Shokman, 2015-11-19:
@voon I used this configuration: https://github.com/PX4/Firmware/tree/master/posix-configs/SITL

Maybe HOST port is inadequate now that you mention it... but it was working, I will check that. 
Also I thought about wireshark, but since it shouldn't be anything sending nothing to that port. Anyway, I can just change the posrt number just in case...

dogmaphobic, 2015-11-19:
> how about on doing multicast UDP? In that case we may omit those bridges completely.

Not sure where to start…

In a *normal* scenario, QGC acts as a *server*. It listens for packets on its UDP port (14550 by default) and *reacts* accordingly. It even registers itself through Avahi/Zeroconf so *clients* can find it. Each *connection* (using that term loosely here as we’re talking about a connectionless link) is handled individually. It means a vehicle is connected, with its unique ID and traffic to and from it is exclusive. Conversation with a vehicle is not shared with other vehicles.

That’s the *normal* scenario. It can however, do some different things. You can enter an URL  (address:port) where it forwards everything it gets. I guess this is where you are thinking of using multicast? I wonder how multiple things, expecting specific responses would react in this scenario.

DonLakeFlyer, 2015-11-19:
Whatever you guys are talking about make sure you take into account the new autoconnect stuff.

DonLakeFlyer, 2015-11-30:
Is there some sort of QGC problem here?

LorenzMeier, 2015-11-30:
No, it works fine for me, closing.

closed 2015-11-30 by LorenzMeier