mavlink/qgroundcontrol #2245: Use AUTOPILOT_VERSION.uid as trigger for user based vehicle identification

DonLakeFlyer, 2015-11-19:
When we see a new one of these we can ask the user to set a unique mavlink vehicle id as well as a user friendly name. Then we can use the name in the vehicle picker dropdown plus te vehicle id which will make things more user friendlty. Probably other places where we use vehicle id for user prompt that should be changed as well. Need to fine those. User friendly vehicle name (user name + id) should come from AutoPilotPlugin call.

LorenzMeier, 2015-11-19:
Requesting AUTOPILOT_VERSION works like this:
http://mavlink.org/messages/common#MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES

DonLakeFlyer, 2015-11-19:
Also uid should be added to Vehicle object.

DonLakeFlyer, 2016-01-21:
Planning on add QGC configurable vehicle information for survey support. Things like what camera you have on the vehicle, turn radius and so forth. We'll also capature a Vehicle Name and ask for ID set along with this.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier