mavlink/qgroundcontrol #2252: Disconnecting one out of multiple vehicles is not quite right.

dogmaphobic, 2015-11-20:
If I have two vehicles connected and then disconnect one (shut it down for instance), the UI doesn't recover from it.

If I disconnect the *Current Vehicle* (the one showing in the tool bar), nothing happens. No messages, no errors, no UI changes. If I however, click the vehicle button and select the other one (the one still connected), the tool bar goes blank (the normal *disconnected* state).

If I disconnect the *Other Vehicle*, the toolbar goes blank (the normal *disconnected* state as above).

From that point, there is no return of the toolbar.

[image]

DonLakeFlyer, 2015-11-20:
Is this on current master? If you are using the AutoConnect pull it was most likely broken by that.

dogmaphobic, 2015-11-20:
It's using your ACWork branch. Let me try master.

dogmaphobic, 2015-11-20:
Master has similar issues. Can't quite tell because the *Vehicle* button doesn't show up at all so there is no way to select the *other* vehicle.

If I disconnect any vehicle, the toolbar goes blank as well even though one of the vehicles is still connected.

dogmaphobic, 2015-11-20:
This code makes the items in the toolbar visible or not:
```
Loader {
    source:             multiVehicleManager.activeVehicleAvailable ? "MainToolBarIndicators.qml" : ""
    anchors.left:       parent.left
    anchors.verticalCenter:   parent.verticalCenter
}
```
What exactly is ```multiVehicleManager.activeVehicleAvailable```? Is it asking if the *active* vehicle is available? Maybe the *disconnected* vehicle is still *active* but obviously not available.

I could dig further but I want to get the map caching code done :P

DonLakeFlyer, 2015-11-20:
> What exactly is multiVehicleManager.activeVehicleAvailable

Basically does what is says. true: there is a vehicle available, false: no vehicle. Kind of the same as multiVehicleManager.activeVehicle == NULL or not. The active vehiclet transition is not working correctly.

tfoote, 2016-11-22:
I've recently run into this using a recent version 3.0.2

When the actively selected vehicle goes away, the toolbar dissapears and the "Disconnect" button appears. The only way to get back to the toolbar is to click "Disconnect" but that will disconnect all the other vehicles still active on the same mavlink connection. 

[image]

DonLakeFlyer, 2017-05-18:
Retested and this is working correctly

closed 2017-05-18 by DonLakeFlyer