mavlink/qgroundcontrol #2262: Windows daily build access denied for installer

LorenzMeier, 2015-11-22:
@dagar The windows daily build is currently not accessible it seems:
https://s3-us-west-2.amazonaws.com/qgroundcontrol/master/release/qgroundcontrol-installer.exe

Is there an issue with the appveyor upload?

dagar, 2015-11-22:
After the switch to shadow builds on travis-ci and appveyor and the cleanup I adjusted the appveyor upload location to be consistent with the others.

It's now https://s3-us-west-2.amazonaws.com/qgroundcontrol/master/qgroundcontrol-installer.exe

I dropped a note in the PR to get it updated on the wiki, but I guess it slipped through.
While we're on the topic would you be able to fix the wiki so I could register? The CAPTCHA is broken.
http://qgroundcontrol.org/start?do=register

DonLakeFlyer, 2015-11-22:
@LorenzMeier I don't know much about the wiki setup? Any help for Daniel so he can update the wiki himself?

DonLakeFlyer, 2015-11-22:
Wiki updated

LorenzMeier, 2015-11-22:
Thanks Don - I probably won't bother to fix the wiki since its pretty broken. Will be replaced within the next days with a real website.

closed 2015-11-23 by DonLakeFlyer