mavlink/qgroundcontrol #2263: Qt Unknown modules on Debian

erikd, 2015-11-23:
Trying to build this on Debian (Testing). Studied the `.travis.yml` file for how to do this, but when it comes to:

```
export CONFIG="debug installer doxygen"
export SPEC=linux-g++-64
qmake -r ${TRAVIS_BUILD_DIR}/qgroundcontrol.pro CONFIG+=WarningsAsErrorsOn -spec ${SPEC}
```

I get:

```
Project ERROR: Unknown module(s) in QT: location-private positioning-private
```

To begin with there were a lot more unknown modules, but installing various packages fixed it. For these two however I have already installed "libqt5location5 libqt5location5-plugins libqt5positioning5 libqt5positioning5-plugins qtlocation5-dev qtpositioning5-dev" and for some reason there don't seem to be any `-private` packages for these in the Debian repos.

Anyone have any idea of how `qmake` detects this? If I can get a file path for missing file I can use `apt-file search` to figure out which package its in.

DonLakeFlyer, 2015-11-23:
Best way is to follow the README: https://github.com/mavlink/qgroundcontrol/blob/master/README.md <https://github.com/mavlink/qgroundcontrol/blob/master/README.md> and use the Online installer to install Qt. Those packages may not necessarily be the right version and as you can se from the unknown modules error, they are missing modules.

DonLakeFlyer, 2015-11-24:
It's not that you are missing packages. It's that the packages are missing the private models. If you install using Qt online installer as per README, everything should work fine.

closed 2015-11-24 by DonLakeFlyer