mavlink/qgroundcontrol #2264: Font issue with daily build under windows

sirPerna, 2015-11-23:
I just tested the daily QGC build under windows 10 64bit. It looks like ther is an issue showing the text correctly.

sirPerna, 2015-11-23:
Here is a picture how it looks like.
[image]

DonLakeFlyer, 2015-11-23:
Is it all fonts size in all location? Or just a specific size or location?

sirPerna, 2015-11-23:
Looks like specific size and location
[image]
I see it on all selected drop downs but when you click on the drop down the other entries are fine. I also see it on the text for the firmware update and where you select the firmware you want to load. Also on all text boxes but only the text inside the text box.

LorenzMeier, 2015-12-10:
@sirPerna Can you re-try the current daily? There was a recent switch in some of the infrastructure code which might have improved things. Quite possible that we fully need to switch to QML on desktop as well to completely fix it though.

DonLakeFlyer, 2015-12-10:
This is some strange thing in Qt. I can report it, but I can't figure out what is going on. It only happen in Combo Boxes in this panel. They are fine other places in the code.

sirPerna, 2015-12-11:
@LorenzMeier Just tested but the issue is still there (v2.8.0-202-g9f86fc0)

DonLakeFlyer, 2016-01-04:
This is a Qt bug. Given the fact that I'm going to re-work this screen post 3.0 release it not worth try to hack around some more to work around the bug. At this point a work around doesn't seem possible. Moving out of the 3.0 release.

NaterGator, 2016-01-08:
This bug is related to the [renderType](http://doc.qt.io/qt-5/qml-qtquick-text.html#renderType-prop) being set to NativeRendering when there are some transformations applied or inherited. 

For example, to fix the Firmware page simply add `renderType: Text.QtRendering` [here](https://github.com/mavlink/qgroundcontrol/blob/0d962e0a66ba6c421362edc2f42b0ef128b3de6a/src/VehicleSetup/FirmwareUpgrade.qml#L378).

I'm unclear on when this default changed / how certain QML components are inheriting NativeRendering, but that's the issue (it is very briefly touched upon [in the docs](http://doc.qt.io/qt-5/qml-qtquick-text.html#renderType-prop)).

NaterGator, 2016-01-08:
Per [this](https://bugreports.qt.io/browse/QTBUG-46030) Qt bug report the issue can also be mitigated by not using fractional positioning of the SetupView.

DonLakeFlyer, 2016-01-10:
@NaterGator Thanks for the fix on this.

closed 2016-01-10 by DonLakeFlyer