mavlink/qgroundcontrol #2266: Android listing new screenshots and description

dagar, 2015-11-23:
- [ ] tablet screenshot
 - [ ] phone screenshot
 - [ ] video?

DonLakeFlyer, 2015-11-23:
I think we need to make a call on minimum screen size that we say we currently support. Do we really want to say we support phones at this point?

dogmaphobic, 2015-11-23:
I can handle the graphics, screen grabs, etc. I'm not going to be the first to volunteer for the copy :)

Regarding sizes, at least on iOS, I limited to iPad for the time being. It's not really usable on smaller screens (phones) until we come up with a reduced functionality UI for them.

DonLakeFlyer, 2015-11-23:
U defaneatly don wont me for the copie!

dagar, 2016-01-06:
@DonLakeFlyer do you have a milestone in mind for a wider android release (and this issue)?

DonLakeFlyer, 2016-01-06:
I'm thinking we go Stable on Android after the 3.0 release. Hoping to get enough APM testers to wring android out some more.

For now I'll grab some more screenshots and clean the store listing up.

dagar, 2016-01-06:
I've started flying tablet only so I can update the screenshots.

I'm somewhat concerned we're going to receive negative reviews from phone users. I wonder if we should limit distribution to only large phones + tablets. Thoughts?

DonLakeFlyer, 2016-01-06:
> I wonder if we should limit distribution to only large phones + tablets. Thoughts?

I put in 7" and above in the text. Could you see if there is some way to set that up so that it rejects devices with smaller screens?

DonLakeFlyer, 2016-01-13:
I just pushed a pile of new screenshots

dagar, 2016-01-13:
New screenshots look good.

@DonLakeFlyer @LorenzMeier @dogmaphobic - I put the current description in my first comment, please edit it in place and I'll update the play store listing.

DonLakeFlyer, 2016-01-13:
Let me look at the things like Tower or DJI description looks like for comparison.

DonLakeFlyer, 2016-01-21:
I updated the description. This should get us far enough for 3.0 release.
------
QGroundControl provides full flight control and configuration for ArduPilot or PX4 powered vehicles. The goal for QGroundControl is improved ease of use for new users as well as high end feature support for experienced users. Support for ArduPilot vehicles is currently in Beta.

QGroundControl features:
- Full configuration support for ArduPilot and PX4 vehicles
- Mission planning for autonomous flight
- Flight map display showing vehicle position, flight track, waypoints and vehicle instruments
- Video streaming with instrument display overlays
- Flight support for any MAVLink capable vehicle
- Competition-proven at IMAV and EMAV competitions

http://qgroundcontrol.org

closed 2016-01-21 by DonLakeFlyer