mavlink/qgroundcontrol #2268: Decide on min fully supported screen size/dpi and so forth for android

DonLakeFlyer, 2015-11-23:
Here is some info: http://developer.android.com/about/dashboards/index.html.

My proposal is that there is a minimum config that we fully support. Anything below that is "maybe it will work" until we create a custom ui for small screens.

dogmaphobic, 2015-11-23:
I've added a (runtime) property to detect *tiny* screens:

```
ScreenTools.isTinyScreen
```

DonLakeFlyer, 2015-11-23:
What's the definition of "tiny"?

dogmaphobic, 2015-11-23:
For now it's hardcoded to anything smaller than 120mm:

```
property bool isTinyScreen: (Screen.width  / Screen.pixelDensity) < 120 // 120mm
```

DonLakeFlyer, 2015-11-29:
I'd like to propose 7" as the lower bound of currently supported tablet sizes. This covers all iPad and most non-phone android tablet I believe.

dogmaphobic, 2015-11-29:
Android manifest has an attribute (*android:requiresSmallestWidthDp*) that defines the minimum supported size. It's in a rather strange unit though:

http://developer.android.com/guide/topics/manifest/supports-screens-element.html

Note that this is only used to filter the app out of the Google Play Store. At runtime, this is ignored. We still need the code above (though updated to 7" or 177.8mm).

For iOS, it's already done. I've set a limit to iPads. It will not install on any iPhone.

DonLakeFlyer, 2015-11-29:
My main concern is to not have people say "Your app sucks" because they are running it on a too small screen size that requires work we haven't put in yet for UI relayout/redesign. We're kind of getting some of that already.

dagar, 2016-01-13:
Android now limited to 7"  https://github.com/mavlink/qgroundcontrol/issues/2551

closed 2016-01-13 by dagar