mavlink/qgroundcontrol #2271: Advanced Mission Item Issue

dogmaphobic, 2015-11-24:
Putting this here so I won't forget it:

If I select a mission command that is not part of the *normal* commands, the UI has issues with it.

* Select a mission item
* Change the drop down to something weird (MAV_CMD_PAYLOAD_PREPARE_DEPLOY for instance)
* Select a different mission item
* Select the mission item you changed above

The combo box is being set to an non-existing item:

```
file:///Volume/Data/local/Qt/5.5/clang_64/qml/QtQuick/Controls/ComboBox.qml:569: ReferenceError: comboBox is not defined
```

And the UI gets messed up (no *CogWheel*)

[image]

DonLakeFlyer, 2015-11-24:
The qt output is actually some sort of strange control bug on the Qt side. Nothing goes wrong because of it. The cog wheel goes away because you have switched an item which has no "friendly editor" so there is nothing to switch to. Same thing can happen if there is a friendly editor, went to raw and then changed to settings to something the friendly editor can't represent. So you can't go back at that point.

DonLakeFlyer, 2015-11-24:
I think there may be something with that combo box warning. I can't quite figure it out yet.

DonLakeFlyer, 2015-11-30:
Still get bogus warning, but nothing else wrong

closed 2015-11-30 by DonLakeFlyer