mavlink/qgroundcontrol #2280: Bad formating of Latitude and or Longitude

Fojtik, 2015-11-24:
I have in LAT displayed a number 137190049871108. Please could you fix double number formating to the left edge.

When LAT contains 51.137190049871108, the number seen in editbox seems crazy.
[image]

DonLakeFlyer, 2015-11-25:
This is a bug with decimal places. Lat/Lon/Alt should be limited to 7 decimals places. Which in turn prevents it from scrolling out of view.

DonLakeFlyer, 2015-11-25:
Tried to repro. Decimal places seem to working correctly. This image looks to be from an older build. I think that was fixed after that. Can you try a new daily?

Fojtik, 2015-11-25:
Where to find daily build?
I has been asked you by Email, but this Email has been dropped and unread.

The daily build is no longer available from here:
http://www.qgroundcontrol.org/downloads

LorenzMeier, 2015-11-26:
Just look again. Its all there and has always been.

Fojtik, 2015-11-26:
Yes, now it works again.

Before while the Windows download has been broken:
<Error><Code>AccessDenied</Code><Message>Access 
Denied</Message><RequestId>1004EAAA5C1A4035</RequestId><HostId>xIs24tYD5CuPFh7NV3V2E96SNCs
XlolYgP82Bn2yGIIZtaumeceq7Z6oNrLLvh+ULnxs0fxzjh4=</HostId></Error>

LorenzMeier, 2015-11-26:
Is the number display fixed for you? Can we close this?

Fojtik, 2015-11-26:
Yes, I do not notify this issue.

closed 2015-11-26 by Fojtik