mavlink/qgroundcontrol #2288: Waypoint operation does not work

Fojtik, 2015-11-26:
Waypoint operation does not work, see attached image.

LorenzMeier, 2015-11-26:
@DonLakeFlyer You have to ensure that you are *only* ever rounding during visualisation. You need to make sure to save / print those numbers with at least 18 digits of precision.

DonLakeFlyer, 2015-11-26:
@Fojtik Can you attach the waypoint file? Also specify the exact set of steps your used to repro this.

@LorenzMeier I'm using 7 digit rounding for display values in the ui and 18 digit file format output precision. I believe this is exactly the same as the old code.

DonLakeFlyer, 2015-11-26:
@Fojtik Also where are you getting this build from? Are you building yourself? Or are you using a daily. If daily, which one The visuals still look like they are coming from an older build.

Fojtik, 2015-11-26:
Before 2 years I can build QGC. Now it is not possible for me to build it.
Waypoint file is a same - AroundCech.waypoint we already discussed.

I have loaded this file and uploaded to my emulator. These two screans appeared.

I can explain more details on Email.

DonLakeFlyer, 2015-11-26:
Use the gitter channel https://gitter.im/mavlink/qgroundcontrol for discussion on build problems.

DonLakeFlyer, 2015-11-26:
This is a bug in master. Stable works, although it doesn't allow you to edit. Which is correct behavior in Stable, since this set of waypoints is using settings which are not supported by visual editing.

The bug is that the values are incorrectly getting coverted back and forth to an int when sent to a vehicle. Introduced by the new raw edit capability just added and a bad copy constructor which wasn't setting the FactMetaData correctly. Unit test didn't catch it because the test values I was using don't have any digits past the decimal point. Fixing that all up now.

DonLakeFlyer, 2015-11-27:
This still leaves the issue that you are using a MAV_CMD=0 which from a mavlink standard standpoint is not really supported. I'd rather not implement a mechanism to support editing of unknown MAV_CMD ids. My suggestion would be to add a couple MAV_CMD_USER1, MAV_CMD_USER2 ids to the mavink spec that can be used by anyone to code custom commands. Then that could be supported in a standard way. Other option is to just take over one of the other real MAV_CMD ids and use that instead of 0.

LorenzMeier, 2015-11-27:
Yes. Lets reserve a range of 20 commands. But it should not be USER1 but suggesting it should never ever be used in a production system, like TESTING1 or DEBUG1. While at it we should establish a range reservation for these. we have an int16 here, so plenty of ranges to reserve.

DonLakeFlyer, 2015-11-27:
I'll put together a mavlink pull for it. Thanks Lorenz.

Fojtik, 2015-11-27:
I can pack "my" waypoint to any MAV_CMD that you recommend. This was only a first attempt to do something.

DonLakeFlyer, 2015-11-27:
Ok. Until I get around to adding the new MAV_TESTING MAV_CMDs tp mavlink and update QGC yo can just pick one of the existing MAV_CMDs to take over for your use. I'd suggest something that doesn't have already have a friendly editor associated with it like say MAV_CMD_DO_SET_PARAMETER. Then you can try this whole thing again using the latest daily build to see if you are still having problems. Let us know how it goes.

DonLakeFlyer, 2015-11-30:
This should bd working now, so closing. Will re-open if there are still problems.

closed 2015-11-30 by DonLakeFlyer

Fojtik, 2015-12-01:
Yes, you are right, current daily snapshot does not suffer with reported issue.

DonLakeFlyer, 2015-12-01:
Great, thanks for reporting back @Fojtik