mavlink/qgroundcontrol #2293: manual control lost and regained

ChristophTobler, 2015-11-27:
when I run a simulation with gazebo and use the default UDP link in qgc, I get "MANUAL CONTROL LOST" and "MANUAL CONTROL REGAINED" all the time. see sceenshot (I use a Logitech joystick and it works from time to time)
[image]

DonLakeFlyer, 2015-11-27:
I"m going to have to debug this through from the QGC to the Firmware side. It seems like the time delta for lost control is way too large. Worried that UDP pipe is delayed somehow.

DonLakeFlyer, 2015-11-27:
@LorenzMeier I instrumented both QGC and Firmware SITL (I used the driver_framework branch) side to output message times. Then tested on mac SITL. QGC is correctly sending MANUAL_CONTROL at about 25hz, whereas Firmware side if only seeing anout 4Hz of MANUAL_CONTROL commands. Do you think you could help out here? Not sure if firmware is tossing some MANUAL_CONTROL messages because it's busy or maybe it's losing UDP packets. Not really sure.

ChristophTobler, 2015-11-30:
@DonLakeFlyer I guess: "Do you think you could help out here?" was also meant for @LorenzMeier.? (not sure because he is not a participant) Or can I help or provide more info?

DonLakeFlyer, 2015-11-30:
Yes, hoping that @LorenzMeier can help to debug from the firmware side of things.

tumbili, 2015-11-30:
We know now what the problem is. Me and @ChristophTobler were working on the same network and so our QGC's would connect to the other one's SITL.

ChristophTobler, 2015-11-30:
True. That was one problem. But the MANUAL_CONTROL lost still exists most of the time which has to do with the lidar/opticalflow messages, I guess.

ChristophTobler, 2015-12-01:
Was publishing on UDP Port 14550 instead of 14556. Problem disappeared. Thanks!

closed 2015-12-01 by ChristophTobler