mavlink/qgroundcontrol #2294: Ver 2.8.0 Won't Run In Win 10, QML Error

stevec5000, 2015-11-27:
Won't run in Windows,

DonLakeFlyer, 2015-11-28:
Any more details than that?

stevec5000, 2015-11-28:
Yes, here is everything I have about it.  Why didn't someone check this before it was released to see if it even worked before releasing it?  It's just as bad now as when it wouldn't run back in 2014!  Mission Planner runs OK but can't calibrate the compass, which is why I'm trying QGC again.    Maybe work should be done on fixing some of the problems rather than creating new ones?

1. When downloading the file for 2.8.0 I got lots of warnings saying "This kind of file can harm your system. Are you sure?"  It must look like a virus or contains viruses.
2. After installing the file and starting it up I got the QML Error message and it wouldn't run.
3. Removed and re-installed again, this time it started up displaying the photo of field without displaying the QML Error but seemed to freeze for a while then said could not connect to Com4, the USB port. Also said "Not Responding" across the bar at the top of the window.
4. Killed it and restarted, this time says across the middle of screen: "Click Connect on Top Right To Fly" but there is nothing at the top right!
5. Selected Firmware Upgrade, unplugged the USB and plugged it back in and got these messages:

QGroundControl can upgrade the firmware on Pixhawk devices, 3DR Radios and PX4 Flow Smart Cameras.
Device must be disconnected from USB to start firmware upgrade. Disconnect Pixhawk from usb.
Plug in your device via USB to start firmware upgrade
Found device: Pixhawk
Error: Sync: Send Command: Invalid sync response: 0x27 0xfe
Upgrade cancelled

GroundControl can upgrade the firmware on Pixhawk devices, 3DR Radios and PX4 Flow Smart Cameras.
Device must be disconnected from USB to start firmware upgrade. Disconnect Pixhawk from usb.
Plug in your device via USB to start firmware upgrade
Found device: Pixhawk
Error: Sync: Send Command: Invalid sync response: 0x3d 0x00
Upgrade cancelled

QGroundControl can upgrade the firmware on Pixhawk devices, 3DR Radios and PX4 Flow Smart Cameras.
Plug in your device via USB to start firmware upgrade
Found device: Pixhawk
Error: Sync: Send Command: Get Command Response: Timeout waiting for bytes to be available
Upgrade cancelled

This is displayed in the DOS box:
I] at :0 - "Logging ini file directory "C:/Users/Steve/AppData/Local/QtProject""
[I] at :0 - "Filter rules "qt.network.ssl.warning=false\n""
[I] at :0 - "Settings location "C:/Users/Steve/AppData/Roaming/QGroundControl.org/QGroundControl.ini" true"
[I] at :0 - "Saved files location "C:/Users/Steve/Documents/QGroundControl""
[I] at :0 - "Mapping cache directory: "C:/Users/Steve/AppData/Local/cache/QGCMapCache55""
[I] at :0 - "0"
[I] at :0 - "Orphaned log file count 0"
[I] at :0 - ""Error connecting: Could not create port.""
[I] at :0 - "QGCMessageBox (unit testing) "Link Error" "Error on link \\\\.\\COM4. Error connecting: Could not create port.""
[I] at :0 - "Mapping cache directory: "C:/Users/Steve/AppData/Local/cache/QGCMapCache55""
[I] at :0 - ""Error connecting: Could not create port.""
[I] at :0 - "QGCMessageBox (unit testing) "Link Error" "Error on link \\\\.\\COM4. Error connecting: Could not create port.""

stevec5000, 2015-11-28:
Also, besides having nothing in the upper right to click and no way to connect, 3 options under Help do nothing and when View, Full Screen is selected there is no mouse or other kind of control available!

DonLakeFlyer, 2015-11-28:
I'm sorry Steve but your are just too much of an asshole for anyone to help you. You should just go back to Mission Planner.

closed 2015-11-28 by DonLakeFlyer

stevec5000, 2015-11-28:
Help me?  How about fixing your crappy program's problems and getting it to work? I've been struggling with this pile of rubbish for one and a half years now and it hadn't worked yet!
Actually you can help me, I figure you guys owe me a new PX4 since I wore out the first one just trying different versions of QGC to see if the hundreds of bug fixes would ever make it work!  At first I though QGC must have broke it but now I think it was just worn out from handling it so much all that time and trying hundreds of different versions of the program and since it was never able to fly that must have been it.