mavlink/qgroundcontrol #2298: Rendering issues on Samsung Note 3

LorenzMeier, 2015-11-28:
The rendering on the Note 3 is way too big and looks like the icon size settings are for a large screen when in facts its a small device (phablet).

Can we please enable the rendering feedback output for all mobile builds in e.g. the new preferences section behind the hamburger? e.g. General, MAVLink (there is a typo there), Developer. We can remove it later, but right now we need our user base to be able to provide us feedback.

That screen should also have instructions how to provide the feedback (best would be a "send screen info to developers" button).

@DonLakeFlyer @dogmaphobic A good feedback process will ensure we get mobile under control.

DonLakeFlyer, 2015-11-28:
Can you add screen shot, also a screen shot of text sizes on other apps? Wondering if the concept of Qt setting a decent default font size just isn't valid on all devices. Maybe we need to go directly to the OS to get a good default font size.

DonLakeFlyer, 2015-11-29:
Just realizing this a 5" tablet. I don't think our current concept of font sizing will work that small.  I really think 7" is a min size for the current ui. Anything below that needs a new ui paradigm.

dogmaphobic, 2016-05-06:
Another which I think has been all resolved by #3333.

closed 2016-05-06 by dogmaphobic